prusa3d/PrusaSlicer #4962: [2.3.0-alpha2+] - Prusament PC Blend profile not available

haplm, 2020-10-24:
### Version
2.3.0-alpha2+

### Operating system type + version
macOS Catalina (10.15.7)

### 3D printer brand / version + firmware version (if known)
Original Prusa MK3S, 3.9.1

### Behavior
* PC Blend isn't available in the selection of filament profiles in the Configuration Assistant, see:
[image]

haplm, 2020-10-24:
I have found an option "Show incompatible print and filament presets" in Preferences. When I tick it, PC Blend is back. It isn't however marked with *, suggesting that it would be incompatible (which is strange, why it isn't shown in the first place then?).

rtyr, 2020-10-24:
This is most likely a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/4895. Restart the PrusaSlicer and click on the notification to update the settings.

haplm, 2020-10-25:
Thanks, that worked. I wasn't ignoring the notification though, it didn't appear until like two restarts...

closed 2020-10-25 by haplm