prusa3d/PrusaSlicer #4967: sequential output breaks sliced visualization

cabbey, 2020-10-25:
### Version
2.3.0-alpha2

### Operating system type + version
Windows 10 Home
[sysinfo.txt](https://github.com/prusa3d/PrusaSlicer/files/5434189/sysinfo.txt)

### 3D printer brand / version + firmware version (if known)
Prusa i3 mk2.5s running previous version of firmware (just got the 3.9.1 upgrade notice today.)

### Behavior
* enabling sequential output breaks slice preview.

[image]

[image]

#### Project File (.3MF) where problem occurs
[bug report example.zip](https://github.com/prusa3d/PrusaSlicer/files/5434209/bug.report.example.zip)

rtyr, 2020-10-25:
Known issue. We are working on that.

[image]

cabbey, 2020-10-26:
ah, I didn't know there was an alpha0 release... had only gone back to look at alpha1 looking for known issues so didn't see those.

closed 2020-10-26 by cabbey