prusa3d/PrusaSlicer #4969: Build failure related to xdg integration

pmjdebruijn, 2020-10-25:
### Version
2.3.0~alpha2+1~g493cdbd06

### Operating system type + version
Ubuntu 20.04 LTS building against system wxGTK 3.0.x

### Behavior
Build fails, with the regression introduced in 057c8a1c1ddaaf94dc7056300f23e326ee08eaaa and reverting the commit results in a successful build again.

It seems the involved bits are only available in the development version of wxGTK, which isn't readily available on any common distro.

```
[ 84%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/MainFrame.cpp.o
cd "slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/obj-x86_64-linux-gnu/src/slic3r" && /usr/bin/c++  -DBOOST_ALL_NO_LIB -DBOOST_ATOMIC_DYN_LINK -DBOOST_CHRONO_DYN_LINK -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_LOCALE_DYN_LINK -DBOOST_LOG_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DLIBNEST2D_GEOMETRIES_clipper -DLIBNEST2D_OPTIMIZER_nlopt -DLIBNEST2D_STATIC -DSLIC3R_GUI -DTBB_USE_CAPTURED_EXCEPTION=0 -DUNICODE -DUSE_TBB -DWXINTL_NO_GETTEXT_MACRO -DWXUSINGDLL -D_FILE_OFFSET_BITS=64 -D_UNICODE -D__WXGTK__ -DwxNO_UNSAFE_WXSTRING_CONV -DwxUSE_UNICODE -I/usr/include/dbus-1.0 -I/usr/lib/x86_64-linux-gnu/dbus-1.0/include -I"slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/src" -I"slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/obj-x86_64-linux-gnu/src/platform" -I"slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/src/clipper" -I"slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/src/polypartition" -I"slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/src/hidapi/include" -I"slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/obj-x86_64-linux-gnu/src/libslic3r" -I"slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/src/libnest2d/include" -I"slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/include" -I"slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/src/miniz" -I"slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/src/glu-libtess/include" -I"slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/src/qhull/src" -isystem /usr/include/eigen3 -isystem /usr/lib/x86_64-linux-gnu/wx/include/gtk3-unicode-3.0 -isystem /usr/include/wx-3.0 -isystem "slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/src/libigl" -isystem /usr/include/OpenEXR  -g -O2 -fdebug-prefix-map=slic3r-prusa-2.3.0~alpha2+1~g493cdbd06=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fext-numeric-literals -Wall -Wno-reorder -pthread -O3 -DNDEBUG -fPIC   -Werror=return-type -Wno-ignored-attributes -Wno-unknown-pragmas -pthread -DOPENVDB_ABI_VERSION_NUMBER=6 -std=gnu++17 -Winvalid-pch -include "slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/obj-x86_64-linux-gnu/src/slic3r/CMakeFiles/libslic3r_gui.dir/cmake_pch.hxx" -o CMakeFiles/libslic3r_gui.dir/GUI/MainFrame.cpp.o -c "slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/src/slic3r/GUI/MainFrame.cpp"
In file included from slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/src/slic3r/GUI/Mouse3DController.hpp:17,
                 from slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/src/slic3r/GUI/GUI_App.cpp:61:
/usr/include/tbb/mutex.h:21:140: note: #pragma message: TBB Warning: tbb/mutex.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.
   21 | #pragma message("TBB Warning: tbb/mutex.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.")
      |                                                                                                                                            ^
slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/src/slic3r/GUI/GUI_App.cpp: In lambda function:
slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/src/slic3r/GUI/GUI_App.cpp:324:25: warning: comparison of integer expressions of different signedness: ‘int’ and ‘size_t’ {aka ‘long unsigned int’} [-Wsign-compare]
  324 |             if (cur_len > longest_sub_str.Len())
      |                 ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~
slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/src/slic3r/GUI/GUI_App.cpp: In function ‘bool Slic3r::GUI::check_old_linux_datadir(const wxString&)’:
slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/src/slic3r/GUI/GUI_App.cpp:381:32: error: ‘class wxStandardPaths’ has no member named ‘SetFileLayout’
  381 |         wxStandardPaths::Get().SetFileLayout(wxStandardPaths::FileLayout_Classic);
      |                                ^~~~~~~~~~~~~
slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/src/slic3r/GUI/GUI_App.cpp:381:63: error: ‘FileLayout_Classic’ is not a member of ‘wxStandardPaths’
  381 |         wxStandardPaths::Get().SetFileLayout(wxStandardPaths::FileLayout_Classic);
      |                                                               ^~~~~~~~~~~~~~~~~~
slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/src/slic3r/GUI/GUI_App.cpp:383:32: error: ‘class wxStandardPaths’ has no member named ‘SetFileLayout’
  383 |         wxStandardPaths::Get().SetFileLayout(wxStandardPaths::FileLayout_XDG);
      |                                ^~~~~~~~~~~~~
slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/src/slic3r/GUI/GUI_App.cpp:383:63: error: ‘FileLayout_XDG’ is not a member of ‘wxStandardPaths’
  383 |         wxStandardPaths::Get().SetFileLayout(wxStandardPaths::FileLayout_XDG);
      |                                                               ^~~~~~~~~~~~~~
slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/src/slic3r/GUI/GUI_App.cpp: In member function ‘void Slic3r::GUI::GUI_App::init_app_config()’:
slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/src/slic3r/GUI/GUI_App.cpp:700:36: error: ‘class wxStandardPaths’ has no member named ‘SetFileLayout’
  700 |             wxStandardPaths::Get().SetFileLayout(wxStandardPaths::FileLayout_XDG);
      |                                    ^~~~~~~~~~~~~
slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/src/slic3r/GUI/GUI_App.cpp:700:67: error: ‘FileLayout_XDG’ is not a member of ‘wxStandardPaths’
  700 |             wxStandardPaths::Get().SetFileLayout(wxStandardPaths::FileLayout_XDG);
      |                                                                   ^~~~~~~~~~~~~~
In file included from slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/src/libslic3r/PrintBase.hpp:14,
                 from slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/src/libslic3r/Print.hpp:4,
                 from slic3r-prusa-2.3.0~alpha2+1~g493cdbd06/src/slic3r/GUI/MainFrame.cpp:16:
/usr/include/tbb/mutex.h:21:140: note: #pragma message: TBB Warning: tbb/mutex.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.
   21 | #pragma message("TBB Warning: tbb/mutex.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.")
      |                                                                                                                                            ^

```

lukasmatena, 2020-10-25:
Thanks. The problem is that some of the functions are only available in wxWidgets 3.1 and not in 3.0... Lovely. There are two solutions:

1. Only do the XDG datadir change with wx>=3.1
2. Read the XDG env variable ourselves without the help from wxWidgets 

I will probably go for number 2.

lukasmatena, 2020-10-26:
This should be fixed by f4311f6. PrusaSlicer now reads the environment variable itself.
Maybe there are other issues with wx3.0, but this particular one should be fixed.
Closing.

closed 2020-10-26 by lukasmatena