prusa3d/PrusaSlicer #4970: Invalid zero value for --default-acceleration when using other acceleration settings

pmjdebruijn, 2020-10-25:
### Version
`2.3.0~alpha0+1188~g12b1de53d`

### Operating system type + version
xubuntu 20.04 LTS

### Behavior
When opening a .3mf saved with a previous git master version of PrusaSlicer, I get this error, and PrusaSlicer will not start:

```
error: Invalid zero value for --default-acceleration when using other acceleration settings
```

In principle any invalid metadata in the 3mf should not result in PrusaSlicer not starting, but it should just be ignored, and start anyway.

As a sidenote, loading the 3mf from within PrusaSlicer works fine regardless.

When unpacking the .3mf, I see this:

```
**grep -i accel Metadata/Slic3r_PE.config**
; bridge_acceleration = 300
; default_acceleration = 0
; first_layer_acceleration = 0
; infill_acceleration = 0
; machine_max_acceleration_e = 5000
; machine_max_acceleration_extruding = 500
; machine_max_acceleration_retracting = 1000
; machine_max_acceleration_x = 500
; machine_max_acceleration_y = 500
; machine_max_acceleration_z = 100
; perimeter_acceleration = 0
```

bubnikv, 2020-12-07:
> When opening a .3mf saved with a previous git master version of PrusaSlicer, I get this error, and PrusaSlicer will not start:

How do you start PrusaSlicer? By drag and dropping the offending 3mf onto PrusaSlicer icon?
Or are you slicing from command line? What is your use case?

pmjdebruijn, 2020-12-07:
When I open a 3mf by doubleclicking, PrusaSlicer simply will not start.

When starting from the command like so, I see the error as initially reported:
prusa-slicer whatever.3mf

Likely relevant:
https://github.com/prusa3d/PrusaSlicer/blob/master/src/PrusaSlicer.cpp#L656
https://github.com/prusa3d/PrusaSlicer/blob/master/src/libslic3r/PrintConfig.cpp#L3487

pmjdebruijn, 2020-12-18:
This is still an issue with 2.3.0~rc1

closed 2026-09-29 by prusaslicer-bot[bot]