prusa3d/PrusaSlicer #4973: boost::filesystem::canonical: No such file or directory

pmjdebruijn, 2020-10-25:
### Version
`2.3.0~alpha0+1188~g12b1de53d` built with `-DSLIC3R_FHS=1`

### Operating system type + version
xubuntu 20.04 LTS

### Behavior
When I start prusa-slicer, it wants the directory prusa-slicer to exist in the current working directory, as I'm getting this error:

```
PrusaSlicer GUI initialization failed: boost::filesystem::canonical: No such file or directory: "CURRENT_WORKING_DIR/prusa-slicer"
```

If I create an empty directory it starts just fine.

Regression introduced in 7531da603bb2d8a29be92695df26625c023927f3 as everything works fine again when this commit is locally reverted.

bubnikv, 2020-10-27:
Would you please retest with the current master?
We hope the issue is solved by creating the missing datadir early.

pmjdebruijn, 2020-10-27:
The problem still exists in current master (2.3.0~alpha2+37~g0798fa818)

Keep in mind it wants a `prusa-slicer` directory in my _current working directory_, **not** my _home directory_.

So it's different depending on where the prusa-slicer binary is started.

JacekJagosz, 2020-11-05:
I can confirm the same thing happening on Solus, also works if you create an empty directory.

lukasmatena, 2020-11-06:
@JacekJagosz Does it also happen with our AppImage or have you built it yourself too?

JacekJagosz, 2020-11-06:
I build it myself, I'm maintainer for Solus.
Only clue I see for this problem is it might be connected to wxwidgets, as I build it with WX_STABLE=1, but otherwise I don't see how I could be the problem.
```
PrusaSlicer GUI initialization failed: boost::filesystem::canonical: No such file or directory: "/home/jacek/prusa-slicer"

(process:171438): GLib-GObject-WARNING **: 09:12:54.377: invalid (NULL) pointer instance

(process:171438): GLib-GObject-CRITICAL **: 09:12:54.378: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(process:171438): Gtk-CRITICAL **: 09:12:54.380: IA__gtk_settings_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed

(process:171438): GLib-GObject-CRITICAL **: 09:12:54.380: g_object_get: assertion 'G_IS_OBJECT (object)' failed

(process:171438): GLib-GObject-WARNING **: 09:12:54.380: value "TRUE" of type 'gboolean' is invalid or out of range for property 'visible' of type 'gboolean'

(process:171438): Gtk-CRITICAL **: 09:12:54.380: IA__gtk_settings_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed

(process:171438): GLib-GObject-CRITICAL **: 09:12:54.380: g_object_get: assertion 'G_IS_OBJECT (object)' failed

(process:171438): Gdk-CRITICAL **: 09:12:54.380: IA__gdk_screen_get_display: assertion 'GDK_IS_SCREEN (screen)' failed

(process:171438): Gdk-CRITICAL **: 09:12:54.380: IA__gdk_display_pointer_ungrab: assertion 'GDK_IS_DISPLAY (display)' failed

(process:171438): Gtk-WARNING **: 09:12:54.380: Screen for GtkWindow not set; you must always set
a screen for a GtkWindow before using the window

(process:171438): Gdk-CRITICAL **: 09:12:54.380: IA__gdk_pango_context_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed

(process:171438): Pango-CRITICAL **: 09:12:54.380: pango_context_set_font_description: assertion 'context != NULL' failed

(process:171438): Pango-CRITICAL **: 09:12:54.381: pango_context_set_base_dir: assertion 'context != NULL' failed

(process:171438): Pango-CRITICAL **: 09:12:54.381: pango_context_set_language: assertion 'context != NULL' failed

(process:171438): Pango-CRITICAL **: 09:12:54.381: pango_layout_new: assertion 'context != NULL' failed

(process:171438): Pango-CRITICAL **: 09:12:54.381: pango_layout_set_text: assertion 'layout != NULL' failed

(process:171438): Pango-CRITICAL **: 09:12:54.381: pango_layout_set_attributes: assertion 'layout != NULL' failed

(process:171438): Pango-CRITICAL **: 09:12:54.381: pango_layout_set_alignment: assertion 'layout != NULL' failed

(process:171438): Pango-CRITICAL **: 09:12:54.381: pango_layout_set_ellipsize: assertion 'PANGO_IS_LAYOUT (layout)' failed

(process:171438): Pango-CRITICAL **: 09:12:54.381: pango_layout_set_single_paragraph_mode: assertion 'PANGO_IS_LAYOUT (layout)' failed

(process:171438): Pango-CRITICAL **: 09:12:54.381: pango_layout_set_width: assertion 'layout != NULL' failed

(process:171438): Pango-CRITICAL **: 09:12:54.381: pango_layout_get_extents: assertion 'layout != NULL' failed

(process:171438): Gtk-CRITICAL **: 09:12:54.381: IA__gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed

(process:171438): Gtk-CRITICAL **: 09:12:54.381: IA__gtk_settings_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed

(process:171438): Gtk-CRITICAL **: 09:12:54.381: IA__gtk_icon_size_lookup_for_settings: assertion 'GTK_IS_SETTINGS (settings)' failed

(process:171438): Gtk-WARNING **: 09:12:54.381: Invalid icon size 6

(process:171438): Gtk-CRITICAL **: 09:12:54.381: IA__gtk_icon_theme_load_icon: assertion 'GTK_IS_ICON_THEME (icon_theme)' failed

(process:171438): Gtk-WARNING **: 09:12:54.381: Error loading theme icon 'dialog-error' for stock: 

(process:171438): Gtk-CRITICAL **: 09:12:54.382: IA__gtk_icon_size_lookup_for_settings: assertion 'GTK_IS_SETTINGS (settings)' failed

(process:171438): Gtk-WARNING **: 09:12:54.382: gtkstyle.c:2358: invalid icon size '6'

(process:171438): Gtk-CRITICAL **: 09:12:54.382: IA__gtk_style_render_icon: assertion 'pixbuf != NULL' failed

(process:171438): GLib-GObject-CRITICAL **: 09:12:54.382: g_object_ref: assertion 'G_IS_OBJECT (object)' failed

[... 26 lines cut ...]
(process:171438): Pango-CRITICAL **: 09:12:54.383: pango_context_set_font_description: assertion 'context != NULL' failed

(process:171438): Pango-CRITICAL **: 09:12:54.383: pango_context_set_base_dir: assertion 'context != NULL' failed

(process:171438): Pango-CRITICAL **: 09:12:54.383: pango_context_set_language: assertion 'context != NULL' failed

(process:171438): Pango-CRITICAL **: 09:12:54.383: pango_layout_new: assertion 'context != NULL' failed

(process:171438): Pango-CRITICAL **: 09:12:54.383: pango_layout_set_text: assertion 'layout != NULL' failed

(process:171438): Pango-CRITICAL **: 09:12:54.383: pango_layout_get_extents: assertion 'layout != NULL' failed

(process:171438): GLib-GObject-CRITICAL **: 09:12:54.383: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

(process:171438): Gdk-CRITICAL **: 09:12:54.383: IA__gdk_screen_get_width: assertion 'GDK_IS_SCREEN (screen)' failed

(process:171438): Gdk-CRITICAL **: 09:12:54.383: IA__gdk_screen_get_width: assertion 'GDK_IS_SCREEN (screen)' failed

(process:171438): Pango-CRITICAL **: 09:12:54.383: pango_layout_set_width: assertion 'layout != NULL' failed

(process:171438): Pango-CRITICAL **: 09:12:54.383: pango_layout_get_extents: assertion 'layout != NULL' failed

(process:171438): Pango-CRITICAL **: 09:12:54.383: pango_layout_get_line_count: assertion 'layout != NULL' failed
fish: 'prusa-slicer' zakończony sygnałem SIGFPE (Floating point exception)
```

pmjdebruijn, 2020-11-06:
I have it with WX 3.1.4 as well, and the commit (7531da6) that introduced this issue, didn't change anything with regard to WX...

kocikdav, 2020-11-17:
Hello, would you please try current master. Hopefully the boost::filesystem:canonical function will not crash anymore.

l29ah, 2020-11-17:
On Tue, Nov 17, 2020 at 08:00:19AM -0800, kocikdav wrote:
> Hello, would you please try current master. Hopefully the boost::filesystem:canonical function will not crash anymore.

Now it works, thank you.

-- 
()  ascii ribbon campaign - against html mail
/\  http://arc.pasp.de/   - against proprietary attachments

bubnikv, 2020-11-21:
Fixed in PrusaSlicer 2.3.0-alpha4. Closing.

closed 2020-11-21 by bubnikv