prusa3d/PrusaSlicer #4975: Problems with cyrillic

AndylgTom, 2020-10-26:
**Version**
PrusaSlicer-2.3.0-alpha2

**Operating system type + version**
Win10 x64

**Behavior**
There are problems with displaying cyrillic in the following windows \ tabs.
**-1-**
[image]

**-2-**
[image]

**-3-**
[image]

And quite a few places where the text in .po is translated, but the program still displays in english... But I think it's just alpha, and then everything will be fixed.

bubnikv, 2020-10-26:
Thanks for the report. Yeah, it is too easy for us to forget to localize some texts. We usually find these issues during the localization phase we do internally for the languages we support.

YuSanka, 2020-12-07:
Thanks for report. There were common localization bugs, not related to the Cyrillic. Bugs are fixed with 4736408a460fbf917978185efc28dc1b49a6ed41 and https://github.com/prusa3d/PrusaSlicer/commit/042bfe6be4602445645cc240457dcc6efb46890a and will be a part of next release.

closed 2020-12-07 by YuSanka