prusa3d/PrusaSlicer #4976: Feauture request: Multiple layer thickness in one model

evert4130, 2020-10-26:
I would love if there were a way to print models with multiple layer thickness. It does not have to be variable thickness as in the prusa-fdm style, but just something to speed up printing in areas where details are not important and then print to a high resolution in other areas. Typically when printing in 0.025 mm layers, I think the support structures at the bottom could be printed in 0.1mm layers and then thinner layers after that.

Some way to define layers for between certain heights eg:

Bottom to 5.8mm=0.1 mm layers

5.8 to 22mm=0.05 mm layers

22 to top=0.025 mm layers

I found another similar feature request, https://github.com/prusa3d/PrusaSlicer/issues/3191#issue-523149933 which would also be good, but it seems like it didn't get any attention in the past year.

bubnikv, 2020-10-26:
SLA or FDM? Because for FDM you can do that.

Area5142, 2020-10-26:
SLA (SL1). The request originated from the SL1 &CW1 forum.

Implemented as layer height range modifiers or better the same logic as used in FDM variable layer height.

P-C-R, 2020-10-26:
+1 on this!

bttguy, 2021-02-05:
Interesting. +1

zakutin, 2026-09-18:
still required