prusa3d/PrusaSlicer #4980: Pause printing

reinaldomcz, 2020-10-26:
### Version
_Version of PrusaSlicer used goes here_
2.2.0 + win64
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10 64bit
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
EENDER3 MARLIM 1.8
### Behavior
* _I am not able to pause the printer to change the filament, the printer does not recognize the command.
I have already uninstalled and installed the softwer again, but it was no use

Gracias 

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2020-10-26:
Is it a slicer problem or a firmware issue? It sounds to me you are having
an issue with your printer firmware.

rtyr, 2020-10-27:
PrusaSlicer 2.2.0 emits M601 command to pause the print. This command is most likely not supported by Ender3 firmware (and by Marlin firmware in general). 

You can most likely solve this issue in the PrusaSlicer 2.3.0-alpha, where you can change the default pause command. I believe the `M25` command will work for you.
https://github.com/prusa3d/PrusaSlicer/releases

@pmjdebruijn @probonopd Can you guys please test the `M25` command on stock Ender3 firmware?  We can then update bundled Ender3 profiles.

[image]

pmjdebruijn, 2020-10-27:
My Ender-3 has been upgraded with an SKR Mini E3 V1.2, so it's not stock anymore. M25 pauses the print just fine on Marlin 2.0, though I don't get a way to resume the print, but that just might be a bug in current Marlin.

reinaldomcz, 2020-10-27:
Ok!

Muito obrigado!

rtyr, 2020-10-27:
M25 works fine on my old delta printer with Marlin 2.0.3
[image]

reinaldomcz, 2020-10-27:
Ótimo

Muito obrigado!

reinaldomcz, 2020-10-27:
Could you send me a print of the command line including the temperature
command.

Thank you!

reinaldomcz, 2020-10-27:
So, the ender must not work

rtyr, 2020-10-27:
It depends on your firmware, you have to try it. Stock Ender3 is likely using older Marlin.

Temperatures are there by default in Marlin 2.0.3.

reinaldomcz, 2020-10-27:
Yes the marlin that came in ender3 is the old one and as far as I know
ender3 is not possible to update Marlin.

neophyl, 2020-10-28:
It should be possible to update the firmware on the Ender 3, its listed as one of the supported printers on the TH3D unified firmware page (which I used on my Cr10S).

probonopd, 2020-10-29:
> @pmjdebruijn @probonopd Can you guys please test the M25 command on stock Ender3 firmware? 

Sorry I don't have access to an Ender 3 currently.

reinaldomcz, 2020-10-29:
I tested the M25 command on my Ender3 and the nozzle did not work
paused on top of the piece and when I placed it to continue printing
the printer nozzle came out breaking the piece on the bed.
I am using google translator to write.
Thank you!

bubnikv, 2020-10-30:
> I tested the M25 command on my Ender3 and the nozzle did not work
paused on top of the piece and when I placed it to continue printing
the printer nozzle came out breaking the piece on the bed.

Maybe you should contact the manufacturer of your printer for help on the M
codes?

bubnikv, 2020-11-21:
Fixed by @pmjdebruijn 
Closing.

closed 2020-11-21 by bubnikv