prusa3d/PrusaSlicer #4981: Cannot zoom properly with retina resolution in 2.3.0-alpha

mathiasvr, 2020-10-26:
### Version
2.3.0-alpha2

### Operating system type + version
macOS Mojave 10.14.6

### Behavior
* _Describe the problem_
When using retina resolution for the 3D scene you cannot zoom when the cursor is placed in an area spanning twice the size of the legend view such as feature type, speed, etc.

* _Steps needed to reproduce the problem_
With retina resolution on mac, place the cursor on left or below the legend and notice how scrolling is active in the legend and not zooming the scene when placed up to 2x the width or height of the legend.

* _Expected Results_
Scrolling should only be active on legend when the cursor is placed in the visible legend area, and otherwise be zooming the scene.

* _Actual Results_
Scrolling affects the legend and not the scene when the cursor is in an area covering twice the visible legend, as marked by the red square area below.

[image]

FidelCapo, 2020-10-27:
You're right. I can reproduce it. Thanks for the report.

FidelCapo, 2020-11-26:
Thank you for report. This issue should be fixed in next version.
Closing.

closed 2020-11-26 by FidelCapo