prusa3d/PrusaSlicer #4982: Spikes on buildplate edges with this gcode file

dahuafschmied, 2020-10-26:
### Version
2.3.0 Alpha2

### Operating system type + version
MacOS 10.13.6

### 3D printer brand / version + firmware version (if known)
-

### Behavior
Spikes on buildplate edges with this gcode file
[image]

Just remove the .txt
[Arsch_Ohren_0.16mm_PLA_.gcode.txt](https://github.com/prusa3d/PrusaSlicer/files/5440193/Arsch_Ohren_0.16mm_PLA_.gcode.txt)

foreachthing, 2020-10-26:
Ein Arsch mit Ohren! :-D Ich hatte Angst die Datei zu öffnen...

The jagged lines are **break lines**, like in CAD. Meaning: the build plate was drawn short and this type of line illustrates that.
I got that on every gcode file I could find ... well, maybe not _every_, but a few.

bubnikv, 2020-10-27:
The break lines are there to indicate that the print bed size is not known.

dahuafschmied, 2020-10-27:
ok. thank you for information

closed 2020-10-27 by dahuafschmied

dahuafschmied, 2020-11-04:
thank you.
fixed in alpha3