prusa3d/PrusaSlicer #4985: PrusaSlicer Bug

Mattmst-ai, 2020-10-26:
### Version
_PrusaSlicer 2.2.0+win64_

### Operating System 

Windows 10 Pro  `64bit`

### 3D Printer 

Prusa I3 MK3

### Behavior
Every time I'm using a modifier "cylinder" the cylinder it coming oval
I really don't know what's happening

* _Actual Results_
Bug ?
[image]
[image]

----------------------------------------------
_Is this a new feature request? - **Yes!**_

Measurement Tool/Ruler it will be very helpful! 

**Thank You**

bubnikv, 2020-10-27:
Please provide a 3MF file. Thank you.

Area5142, 2020-10-28:
For measurement Tool/Ruler take a look at issue #3651

bubnikv, 2020-12-07:
Did you apply anisotropic scaling to the object? Because the modifiers are stretched by the same anisotropic scaling as well.

rtyr, 2021-01-28:
No response. Likely anisotropic scaling applied. Closing.

closed 2021-01-28 by rtyr