prusa3d/PrusaSlicer #4986: Remaining print time and SD-percentage calculation not correct

1WileCoyote, 2020-10-26:
### Version
2.3.0 alpha 2

### Operating system type + version
win10 x64

### 3D printer brand / version + firmware version (if known)
Prusa I3MK3S+MMU2S fw 3.9.1.3518 + 1.0.6

### Behavior
Remaining print time and percentage calculation does not work correct.
The SD-percentage goes to 100% and remaining time to 0 too soon.
I sliced a box with a height of 57mm.
When printing at a height of 49.1mm the percentage goes to 100% and remaining time to 0.
I compared the gcodes from 2.2.0 and 2.3.0 alpha 2 with notepad++ and the M73 command sets P to 100 and R to 0 too soon while prusaslicer 2.2.0 sets P to 91 and R to 13 at the same height.
gcodes attached.
[image]
[gcodes.zip](https://github.com/prusa3d/PrusaSlicer/files/5441913/gcodes.zip)

bubnikv, 2020-10-27:
Thanks, we received multiple similar reports internally. @enricoturri1966 is looking into it.

vintagepc, 2020-10-28:
Can confirm as well - Finished a 10h print the other day, screen said 100% / 0:00 but it continued printing for at least another 30 minutes after I noticed that.

(Posting so I get notified when there's activity/a fix)

fe60, 2020-10-29:
@vintagepc 
It's not required to add a comment to get notifications. Just use the "Subscribe" button on the right side of the initial posting.

THiNK3dAUS, 2020-11-01:
Also getting this issue with 2.3a2 (also had it with a1). Loaded a file in pronterface, and looks like a 22min diff between estimated times!
[image]

bubnikv, 2020-11-04:
It will be fixed in PrusaSlicer 2.3.0-alpha3. Closing.

closed 2020-11-04 by bubnikv