prusa3d/PrusaSlicer #4989: Feature Request: Add loading models in STEP format.

Sembazuru, 2020-10-27:
### Version
Version 2.3.0-alpha2 + win64

### Operating system type + version
Windows 10 Pro Version 2004

### 3D printer brand / version + firmware version (if known)
Original Prusa MK3S fw MK3S-3.9.1-3518-Final

### Behavior
This is, I realize, a big feature request. Definitely not something to expect by 2.3.0 final. ;-)

Slicing solid object STEP files would allow more dimensionally accurate prints without huge meshes from decimation of curves. By using the G2/G3 commands on real arcs, holes can be printed more accurately to the design without having to accommodate holes getting smaller by being printed with chord lines (or getting larger if the solid to mesh algorithm uses tangent lines). Complex curves can be smoother being estimated with arcs and straight lines than with straight lines alone.

There should probably be a switch in the printer profile to turn off G2/G3 gcodes for those printers that don't support them.

#### Project File (.3MF) where problem occurs
N/A

foreachthing, 2020-10-27:
Since most printers still travel in lines, one needs to tessellate the model(s).
Whether you have a STEP or STL, doesn't matter. If you could import a STEP file, you'd have to have all those tessellation settings in PS as well or leave it up to PS (and chance) to triangulate your model. _(In my FEM software, I have 5 different options to tesselate my STEP files with!)_
I'm pro STEP in any case (better, more reliable export from CAD and small in size), but I don't think it's a thing for PS - just yet.

Question: G2/G3, can that handle splines, parabolas and ellipsis? And then, how would I know if my printer supports G2/G2 ... there's no such sticker on the frame ;-)

Sembazuru, 2020-10-27:
I've wanted direct STEP import for a while, but what finally prompted me to request this (admittedly long term wish) is the announcement of the Arc Welder plugin for Octoprint:
https://plugins.octoprint.org/plugins/arc_welder/?utm_source=octoprint&utm_medium=announcements&utm_content=1.4.2

But I worry that interpolating an interpretation of a curve would result in less dimensional accuracy. (Sort of like running text through several translators...)

The author of this plugin has some notes for various firmwares, including the Prusa fork to Marlin. He even mentions that he has submitted a pull request in the Prusa firmware repository (and in the comments there is even a throw-away comment from leptun about how it would be nice for slicers to support STEP).

Regarding tessellation of a STEP model for printing, the question is when the tessellation should happen... During import for the internal data model of the object, or only when slicing and calculating gcode moves? I would argue for the second which is probably another reason for this to be a long-term wish. But I also now worry about the .3mf format. Does that format support embedded solid objects in addition to mesh objects? I really don't know much about .3mf. Is it actually strictly a mesh format with meta data for settings, or is it (like avi) just a container format where the actual model can by any data format?

To answer your last question, I'm not 100 percent sure but I think G2/G3 can only handle circular arc segments, not anything fancier. But with clever interpolation one should be able to better approximate splines, parabolas and elipsis with a combination of arc segments and straight lines instead of just straight lines only. And G2/G3 has been a part of Marlin for ages even if some forks don't support it. Which is why I thought that a configuration switch in the printer profile to turn off G2/G3 (call is gcode arcs) support if the target printer doesn't support it. All current model Prusas should... One technique the author of the plugin mentions for checking compatibility is to send and empty G2 and/or G3 command to the printer in a terminal interface. If the response is something to the effect of unknown or unsupported command your printer doesn't support the command. Also, apparently (I haven't tried it yet) on Marlin 2.0.6 and up the M115 command should tell you if it has Arc Support.

ManuelMcLure, 2020-10-27:
Marlin also supports Bezier curves with `G5` if `BEZIER_CURVE_SUPPORT` is enabled, but I don't think it has been heavily tested so I don't know that it's ready for production use.

uis246, 2022-01-31:
Want. Maybe also add CSG support.

lukasmatena, 2022-08-15:
PrusaSlicer 2.5.0-beta1 allows to import STEP files. That fulfills the feature request in the title.

The "do the whole slicing analytically" request is super huge and it is not coming anytime soon. I personally don't think that slicing triangle meshes is what's currently holding 3D printing back.

n8bot, 2022-08-15:
> [...] I personally don't think that slicing triangle meshes is what's currently holding 3D printing back.

There are other things holding FFF back (trapezoidal velocity profiles), but after that is solved then IMO the mesh triangulation will need to be addressed -- not exactly for accuracy/resolution. We can already slice an insanely dense mesh -- instead, the problem will become that our printers will be able to smoothly accelerate to such high speeds (without ringing) that the tiny little MCU will be swamped with too many gcodes.

I have a hunch that bezier splines, curves, or something similar *might* help with that but tbh I have no conclusive proof. It might be better to just stick with meshes and put much more CPU power and RAM into the controller boards.

But thank you for adding STEP! And thank you for using OpenCASCADE.

rom1dep, 2022-08-15:
> I personally don't think that slicing triangle meshes is what's currently holding 3D printing back.

I'm a nobody/newcomer in this field, so take that as you like :) , but with a background in CAM and some experience around machine-tools, I was very surprised to discover that this isn't the norm here in FFF, where much of the fun comes from printing at all scales and repurposing what others have put online at printables or thingiverse.

At least STEP support catching-up is a step (uhh) in the right direction, so, kudos for that!
(now I'm heading to opening an issue for my freecad-generated stepfile which prusaslicer cannot open)

RobinWhitfield, 2022-08-19:
G2 and G3 wwould be a lovely next step for general printer support to go along with STEP, as combined they can drastically decrease file sizes (for some models) for both models and G-code.

If your STL is a high enough triangle count then the STEP side of things matters less for the resulting G-code, as the slicer will turn the stepped curves into curvy G-code moves anyway, but most shared models are not a very high triangle count.

foreachthing, 2022-08-19:
> G2 and G3 would be a lovely next step for general printer support

And in the meantime, you can use this: https://github.com/fieldOfView/Cura-ArcWelderPlugin
Same as Cura uses. Only drawback: you have to use the post-processing section in PS and define the parameters "manually*.
Just tried it with our Ultimaker 2 and it works great!

lukasmatena, 2023-10-23:
The STEP file format import has been implemented in 2.5.0. Support for G2 and G3 G-codes has been implemented in 2.7.0-alpha1. The rest of the discussion is about analytic slicing and as such, it would better fit in #11501. Closing this one.

closed 2023-10-23 by lukasmatena