prusa3d/PrusaSlicer #4990: Feature request: Add a button to the About window to put version information in the clipboard to make bug reports easier

Sembazuru, 2020-10-27:
### Version
Version 2.3.0-alpha2 + win64

### Operating system type + version
Windows 10 Pro Version 2004

### 3D printer brand / version + firmware version (if known)
Original Prusa MK3S fw MK3S-3.9.1-3518-Final

### Behavior
Please add a button to the About window that would put the program version information into the system clipboard. The people submitting bug reports can use this to get fully detailed version information to help the developers identify exactly which codebase is responsible for the user's issue. (Or at the very least have the version string in the About window selectable text so one can clipboard copy-paste it...)

I got this idea from KiCad where the version information copied to the clipboard also includes some build settings to assist the developers with diagnostics. See this screenshot of the KiCad as an example:
[image]
When clicking the "Copy Version Info" button, the text on the button changes to "Copied..." and the following is put on the system clipboard for the version that I have installed:
> Application: KiCad
> Version: (5.1.7)-1, release build
> Libraries:
>     wxWidgets 3.0.5
>     libcurl/7.71.0 OpenSSL/1.1.1g (Schannel) zlib/1.2.11 brotli/1.0.7 libidn2/2.3.0 libpsl/0.21.0 (+libidn2/2.3.0) libssh2/1.9.0 nghttp2/1.41.0
> Platform: Windows 8 (build 9200), 64-bit edition, 64 bit, Little endian, wxMSW
> Build Info:
>     wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8)
>     Boost: 1.73.0
>     OpenCASCADE Community Edition: 6.9.1
>     Curl: 7.71.0
>     Compiler: GCC 10.2.0 with C++ ABI 1014
> 
> Build settings:
>     USE_WX_GRAPHICS_CONTEXT=OFF
>     USE_WX_OVERLAY=OFF
>     KICAD_SCRIPTING=ON
>     KICAD_SCRIPTING_MODULES=ON
>     KICAD_SCRIPTING_PYTHON3=OFF
>     KICAD_SCRIPTING_WXPYTHON=ON
>     KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
>     KICAD_SCRIPTING_ACTION_MENU=ON
>     BUILD_GITHUB_PLUGIN=ON
>     KICAD_USE_OCE=ON
>     KICAD_USE_OCC=OFF
>     KICAD_SPICE=ON

Note, with the build environment the KiCad team uses for packaging for Windows the platform version always seems to be reported as Win 8, even though the platform might be Win 10... The Linux and OSX builds apparently report the platform properly.

#### Project File (.3MF) where problem occurs
N/A

lukasmatena, 2020-10-27:
This is basically what **About->System Info** dialog is for.
You may be right that if it was accessible from About dialog too, it might be a bit more discoverable. I don't know.

Sembazuru, 2020-10-27:
I didn't even notice the System Info dialog and the new issue template doesn't mention it. All the new issue template says for getting the version info for release versions is
> _Use `About->About PrusaSlicer` for release versions_

The problem with the About PrusaSlicer dialog for this use is the version information isn't selectable for copying to the clipboard. Maybe all that needs to happen is the new issue template here on GitHub needs to change to direct users to use the System Info dialog instead of the About PrusaSlicer dialog.

bubnikv, 2020-12-07:
And the error dialogs as well.

bubnikv, 2020-12-12:
Implemented in PrusaSlicer 2.3.0-beta3.

closed 2020-12-12 by bubnikv