prusa3d/PrusaSlicer #4991: RepRapFirmware G10 support automatic start gcode duplicates temperature setting

ardenpm, 2020-10-27:
### Version
2.3.0-alpha2

### Operating system type + version
_Windows 10_

### 3D printer brand / version + firmware version (if known)
Not printer specific.

### Behavior

When the start gcode does not contain an M104 or G10 temperature setting command then PrusaSlicer will automatically add the appropriate variant of the command based on the firmware flavour. However it now seems to add multiple temperature setting and wait commands for the G10 variant. For example, if you replace the start Gcode for a printer (RepRapFirmware Firmware flavour) with:

```
-- Start Gcode
```

Then you will get a snippit like this in the generated Gcode.

```
M107
G10 S220 P0 ; set temperature
G10 S220 P1 ; set temperature
;TYPE:Custom
-- Start G-code
G10 S220 P0 ; set temperature
M116 ; wait for temperature to be reached
G10 S220 P1 ; set temperature
M116 ; wait for temperature to be reached
G21 ; set units to millimeters
```

Additionally it is also interleaving the set and wait for temperature commands, so in effect it will heat up the tools one at a time. This may or may not be the desired default behaviour.

This issue can be worked around by providing your own start gcode and explicitly setting the temperature with G10, the issue applies only to the automatically generated start gcode.

It is likely an issue with the pull request #4553 which I created so I will do an initial investigation. I thin it is because it is adding the set and wait separately for the G10 variant.

ardenpm, 2020-10-27:
Ok, looking at the code it seems setting the temperature before and after the custom Gcode.cpp is by design. First it sets it without the wait and then with the wait. The comment explicitly says before and after.

```
// Set extruder(s) temperature before and after start G-code.
this->_print_first_layer_extruder_temperatures(file, print, start_gcode, initial_extruder_id, false);

...

this->_print_first_layer_extruder_temperatures(file, print, start_gcode, initial_extruder_id, true);
```

With non G10 firmware it uses just M104 when not waiting and M109 when waiting where as the G10 variant uses G10 alone if not waiting and G10 with a separate M116 to wait.

Will keep reviewing, these notes are primarily just for me to keep track of findings.

bubnikv, 2020-10-27:
BTW I refactored your pull request
with e1f6bb4a92e3efd6f477038a0618628c480ea9e7, but I don't think I broke
anything. Your report seems to be independent of that change of mine.

The "standard" way of controlling the extruder temperature by PrusaSlicer
if there is no extruder temp G-code found in the custom start G-code is the
following:

1) Set the target temp at all extruders, don't wait.
2) Insert the custom G-code.
3) Set the target temp of all extruders, wait until the temp is reached.

You will surely find multiple issues at this issue tracker complaining
about this default behavior.

ardenpm, 2020-10-27:
Yes, after looking through the code I see it is intentional and obviously it does no harm to set the same temperature twice (and now that it detects G10 in custom start Gcode it can be used to override the behaviour anyway). I guess it just looks a bit strange with G10 since there is no heat and wait Gcode with G10 and needs the two lines.

I'm still trying to decide if anything should change knowing this. Right now I would say likely not. I guess M109 in Marlin for multiple tools is also going to cause it to heat one tool at a time by default? I'm inclined to say if people want to heat the tools in parallel they should use custom start Gcode for that.

bubnikv, 2020-10-27:
I would say the current code is structured to heat up all the extruders in
parallel already. It first commands all the extruders to start heating up,
then it performs the user block and after that it waits until the last
extruder heats up to its target temperature.

ardenpm, 2020-10-27:
Isn't M109 on Marlin blocking though? So if you have multiple M109 lines like so:

```
M109 T0 S220
M109 T1 S220
```

Then won't execution of the gcode pause after the first line until the temperature is reached? For RepRapFirmware I believe the situation would be the same, with the current implementation the wait for multiple tools will come out something like this:

```
G10 P0 S220
M116
G10 P1 S220
M116
```

This is basically the same, M116 without any parameters waits for all heaters (actually all slowly changing values is how it is defined) to reach their targets. I believe that is also blocking so it would heat them one at a time. I think usually there you'd call the temperature settings commands in a sequence then use a single M116 to wait for them all.

Having said this, I haven't physically tested on the device to see what happens.

ardenpm, 2020-10-27:
Also it looks like if ooze prevention is turned on the automatically generated start Gcode does not raise the temperature to the active printing value before starting the print. If that's not intended then I would assume that is a separate issue.

lukasmatena, 2020-10-27:
> Then won't execution of the gcode pause after the first line until the temperature is reached? 

Probably. But that's why there are the `M104` calls before. The second nozzle should keep heating up while waiting for the first one. At least I believe that's how it was intended. I don't think you can do it differently if you really want to wait for everybody (unless you have a special gcode that does that, such as the M116).

> if ooze prevention is turned on the automatically generated start Gcode does not raise the temperature 

You are probably right. In both that it happens and that it is a separate issue.

ardenpm, 2020-10-27:
Makes sense now (as well as why there are the two temperature calls. I will close this issue. For the ooze one, I'll do further testing and report separately.

closed 2020-10-27 by ardenpm