prusa3d/PrusaSlicer #4992: Slicer image not good

manurafal, 2020-10-27:
### Version
2.3.0 Alpha 2 win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro

### Behavior
[image]

[image]

[image]

When I slicer the resulting image it doesn't look good.

rtyr, 2020-10-27:
Please provide a 3MF project file.

manurafal, 2020-10-27:
[phone_stand-ANDROID.zip](https://github.com/prusa3d/PrusaSlicer/files/5444777/phone_stand-ANDROID.zip)

rtyr, 2020-10-27:
That's STL file, please attach your project 3MF file (it contains your settings, etc.).

[image]

manurafal, 2020-10-27:
[phone_stand-ANDROID_3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5447907/phone_stand-ANDROID_3mf.zip)

rtyr, 2020-10-27:
Thank you. I managed to find the cause of the issue.

Your "Before layer change G-code" contains `HEIGHT:` comment, which apparently confuses our g-code preview (we are using the same comment for layerheight info). I suggest to change `Z HEIGHT:` to `Z:`, it will work fine then.

[image]
[image]

This will work:
[image]

bubnikv, 2020-10-27:
Thanks. We need to improve our parser of our special annotations to be more specific.

manurafal, 2020-10-28:
I've done what you told me and it works.  Thank you.

[image]

closed 2020-10-28 by manurafal

bubnikv, 2020-10-29:
Fixed with d2e5be89e3460964fb3c38db066a21020de0f0db by tightening the rules to parse PrusaSlicer's own G-code annotations.
PrusaSlicer 2.3.0-alpha3 will process such custom G-code correctly.