prusa3d/PrusaSlicer #4993: 97MB professionnal scanned 3D object, STL File cannot be opened

KeepYourVoltage, 2020-10-27:
### Version
_Version of PrusaSlicer used goes here_
 - PrusaSlicer
Version:   2.1.0+win64
Build:     PrusaSlicer-2.1.0+win64-201909160915

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19041), 64-bit edition
Total RAM size [MB]: 34,254MB
OpenGL installation
GL version:   4.6.0 NVIDIA 442.50
Vendor:       NVIDIA Corporation
Renderer:     Quadro K4000/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gl_spirv
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
[... 313 lines cut ...]
### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
  - Prusa i3 MK3

### Behavior
* _Describe the problem_
- I will open a 97MB STL File of a professional 3d-scanned file with PrusaSlicer but nothing happens.
The file seems to be opened, but the virtual bed is still empty.
If I open the same file with Sli3r Prusa Edition 1.41.1 it opens and I can see the object.
If I will slice this object in Slic3r Prusa Edition 1.41.1 it crashes.
What is the problem?

* _Steps needed to reproduce the problem_
- open a big? STL file

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

pmjdebruijn, 2020-10-27:
Is this reproducable using PrusaSlicer 2.2? (or 2.3 alpha even)?

The size of the STL might not matter, and whether it's "professionally" scanned doesn't say anything either.

It's more likely that the STL has some slight error (which is fairly common), and Prusa Slicer can cope with many small errors, but it without a shared sample to reproduce these issues are often hard to solve.

As a workaround, you could open the STL in Blender, and apply a decimate modifier (for some models you can go as low as 0.1 without any noticable loss in print quality, but you'd have to experiment). As far as I'm aware the decimate modifier often resolves errors as essentially a new mesh is generated.

bubnikv, 2020-12-07:
We cannot say much without the model. Please provide the model.

KeepYourVoltage, 2020-12-07:
Hi,

we solved the problem.
I opened it in blender and saved it again in another size.

Thanks for your help.

 

 

Viele Grüße aus Einbeck | Kind regards
Richard Spanagel
_________________________________

 
Infos & Shop:  <http://www.7-lights.de/> WWW.7-LIGHTS.DE

7-Lights – Stefan Rieger / Richard Spanagel GbR
Billerbecker Straße 29
37574 Einbeck

Tel.: 05563/9607145
Mail:  <mailto:info@7-lights.de> info@7-lights.de

StNr.: 12/230/06600
USt.: DE316167917

Du findest uns auch auf diesen Plattformen.
You can find us on these platforms.

bubnikv, 2020-12-07:
thanks, closing.

closed 2020-12-07 by bubnikv