prusa3d/PrusaSlicer #4995: 2.3.0 Alpha 2 - Insert pause uses M601 not the GCode set in the pause print Gcode printer settings

yavinmaster, 2020-10-27:
### Version
_Version of PrusaSlicer used goes here_ 2.3.0-alpha2+win64

### Operating system type + version
_What OS are you using, and state any version #s_ Windows 10

### 3D printer brand / version + firmware version (if known)
Marlin 2,???

### Behavior
* _Describe the problem_

Slice a model.  Go to preview.  Move layer height to desired location.
Right click on + next to layer height selector.

Pause print option is stated as M601 but in Printer settings, Custom G-code, Pause print gcode, is set to something else (M0 in my case).  Confirmed, selected pause print does insert M601 not what is in the customer g-code.  Can obviously use insert custom gcode but this seems to negate the pause print gcode section in the printer setup.

_Is this a new feature request?_ N

lukasmatena, 2020-10-27:
I cannot reproduce that. Could you please send a 3MF? Are you sure you're using the printer profile you're editing?

yavinmaster, 2020-10-27:
Looks like it is my mistake - I don't know why, but the profile change I
made doesn't appear to have taken place.  I can confirm the pause print
function does indeed insert the correct gcode.  What I need to investigate
is why the changes I did to the profile didn't apply (even though I saved
them).  I will open a new report if I can reproduce it.
This report can be closed.

lukasmatena, 2020-10-27:
Ok, no problem, open a new issue if you can reproduce it. I'm closing this one.

closed 2020-10-27 by lukasmatena