prusa3d/PrusaSlicer #4996: Custom profiles prevent the slicer from starting up

io7m, 2020-10-27:
### Version
2.2.0

### Operating system type + version

```
PrusaSlicer
Version:   2.2.0
Build:     PrusaSlicer-2.2.0+UNKNOWN

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 5.8.10-arch1-1 x86_64
Total RAM size [MB]: 33,652MB
OpenGL installation
GL version:   4.6 (Compatibility Profile) Mesa 20.2.0
Vendor:       X.Org
Renderer:     AMD Radeon RX 5700 (NAVI10, DRM 3.38.0, 5.8.10-arch1-1, LLVM 10.0.1)
GLSL version: 4.60
Installed extensions:

GL_AMD_conservative_depth
GL_AMD_depth_clamp_separate
GL_AMD_draw_buffers_blend
GL_AMD_framebuffer_multisample_advanced
GL_AMD_multi_draw_indirect
GL_AMD_performance_monitor
GL_AMD_pinned_memory
GL_AMD_query_buffer_object
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_stencil_export
GL_AMD_shader_trinary_minmax
GL_AMD_texture_texture4
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ANGLE_texture_compression_dxt3
GL_ANGLE_texture_compression_dxt5
GL_APPLE_packed_pixels
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
[... 302 lines cut ...]
getdents64(5, 0x56203471c290 /* 2 entries */, 32768) = 48
getdents64(5, 0x56203471c290 /* 0 entries */, 32768) = 0
close(5)                                = 0
getcwd("/home/rm", 1024)                = 9
stat("/home/rm/.PrusaSlicer/printer", {st_mode=S_IFDIR|0755, st_size=4096, ...}) = 0
lstat("/", {st_mode=S_IFDIR|0755, st_size=4096, ...}) = 0
lstat("/home", {st_mode=S_IFDIR|0755, st_size=4096, ...}) = 0
lstat("/home/rm", {st_mode=S_IFDIR|0711, st_size=4096, ...}) = 0
lstat("/home/rm/.PrusaSlicer", {st_mode=S_IFDIR|0755, st_size=4096, ...}) = 0
lstat("/home/rm/.PrusaSlicer/printer", {st_mode=S_IFDIR|0755, st_size=4096, ...}) = 0
openat(AT_FDCWD, "/home/rm/.PrusaSlicer/printer", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 5
fstat(5, {st_mode=S_IFDIR|0755, st_size=4096, ...}) = 0
getdents64(5, 0x56203471c290 /* 3 entries */, 32768) = 96
stat("/home/rm/.PrusaSlicer/printer/i3_mk3s_1mm_brass_nozzle.ini", {st_mode=S_IFREG|0644, st_size=3668, ...}) = 0
openat(AT_FDCWD, "/home/rm/.PrusaSlicer/printer/i3_mk3s_1mm_brass_nozzle.ini", O_RDONLY) = 6
read(6, "# generated by PrusaSlicer 2.2.0"..., 8191) = 3668
read(6, "", 8191)                       = 0
close(6)                                = 0
getdents64(5, 0x56203471c290 /* 0 entries */, 32768) = 0
close(5)                                = 0
^C--- SIGINT {si_signo=SIGINT, si_code=SI_KERNEL} ---
strace: Process 2353835 detached
```

I haven't read the slicer code, but I'm guessing that it reads in all of the `.ini` files and then handles inheritance afterwards. I'm guessing there's some kind of infinite loop happening in this stage.

io7m, 2020-10-27:
To clarify, the `SIGINT` at the end of the strace output is me giving up and using `^C`.

If I take the files back out of the filament directory, the slicer starts up without issue.

bubnikv, 2020-12-07:
You inherited a profile from itself. I have added a check for that, so the application will just ignore such "inherits" field.
780418435a58fc0367554db0df912cde37acc29f
Closing.

closed 2020-12-07 by bubnikv