prusa3d/PrusaSlicer #4997: Prusaslicer 2.3.0-alpha2 Adding objetcs in mm detected in inches

tazlewouaf, 2020-10-27:
### Version
2.3.0-alpha2

### Operating system type + version
Windows 10 Pro version 2004 (19041.572

### Behavior
when I add objects, PrusaSlicer indicates that some elements are in inches and suggests me to convert them while these elements are in mm (created from Fusion 360)
_Is this a new feature request?_

[image]
[image]

lukasmatena, 2020-10-27:
PrusaSlicer has no way to know what units was the STL created in, this information is not there. It can only guess, and it guesses based on volume of the object. If it is too small, it shows the dialog that user may want to convert it. So there are two possible reasons for it: either your object is really small (is it?), or the detection malfunctioned (maybe because of a broken model or whatever). Can you share the stl?

tazlewouaf, 2020-10-27:
Indeed, it concerns small parts of the whole set. But I also notice that the "inches" parameter is active by default after loading the objects. Wouldn't it be more relevant to disable this default setting and put it in the menu : "Configuration/Preferences" ?

fcore117, 2020-10-30:
Yes, mm should definitely be only default as it is scientifically right parameter.

murk-sy, 2024-07-07:
Bumping this a bit

I'm working with a customer who regularly works with stls in meters, and sometimes instead of "this is made in meters" I get the "this is made in inches" inexplicably.

It's not really the end of the world but it's annoying to have to have some models in 100000% scale just because inches have the audacity to still exist

Ideally,
- Both "recalculate as inches" and "recalculate as meters" are shown regardless of the scale PS determines it to be during import, or
- option whether to ask for any of those is in the settings

First one is likely the easiest to implement and shouldn't be *too* much work, code around here:
https://github.com/prusa3d/PrusaSlicer/blob/fd02400245df25f1286059aa642739660df76663/src/slic3r/GUI/Plater.cpp#L1349

murk-sy, 2026-09-24:
Reproduced in 3.0.0-alpha12

I reimported some models made in meters and they were still detected as in inches.

Ideally, when a size issue is detected, it should just provide options to pick from:

[image]

Obviously the pre-selection here one is incorrect, but at least I can correct it before import in cases when it doesn't work, to avoid having to work with 100000% scale