prusa3d/PrusaSlicer #4998: Supports that rest on an angled surface of the part don't work correctly (sometimes)

spiff72, 2020-10-27:
### Version
About-PrusaSlicer reports version 2.2.0, but I downloaded and installed version 2.2.9.1 (not sure why the about screen doesn't match the version shown on the download page).

### Operating system type + version
Windows 10 - Version 1903

### 3D printer brand / version + firmware version (if known)
Prusa MK3S-MMU2S
Firmware 3.9.1-3518 (from support screen on printer)
MMU2 Firmware 1.0.6-372 (from support screen on printer)

### Behavior
* _Describe the problem_
Slicing certain models with supports enabled will create strange "floating" support structures that don't rest on the surface of the part.  Images and an example 3MF file are included.  When this happens, the supports don't print correctly (since they are printing in mid-air).  Note:  One could argue that this example part probably doesn't actually need supports, but this was just an example where it came up for me recently (and I have seen this behavior in the past as well).

* _Steps needed to reproduce the problem_
Slicing certain models that have a steep angled surface on the part that also has structures ABOVE that angled surface that the slicer thinks are in need of support.

* _Expected Results_
I expect that the support structures should extend all the way down the the angled surface of the part, rather than floating above it.
* _Actual Results_
The bottoms of the support features are suspended mid-air and don't print as they should. 

In the attached image, the supports on the left (circled) are the worst offenders.  I attempted to create a simple demo part that would show this (rather than using the larger model I recently had this issue on), and it has a bit of the issue (circled on the right), but it isn't as bad as the other part.

[image]

_Is this a new feature request?_
NO 

#### Project File (.3MF) where problem occurs
[angled_support_example_102720a.zip](https://github.com/prusa3d/PrusaSlicer/files/5446111/angled_support_example_102720a.zip)

bubnikv, 2021-02-24:
Duplicate of Supports are placed in air #3264

closed 2021-02-24 by bubnikv

HiLivin, 2021-09-19:
The issue seems resolved in 2.4.0-alpha1 with "Snug" support style chosen.
The "Grid" style still exhibits the same problem as before.