prusa3d/PrusaSlicer #5000: Ironing at the end after finishing multipe objects

Vossberger, 2020-10-27:
Version
2.3.0 Alpha-1

Operating system type + version
Windows10 64bit

3D printer brand / version + firmware version (if known)
Mk3s FW3.9.1

Behavior
Last layer including Ironing is performed and causes underextrusion to the next object when multiple same objects are printed. Slicer works from object to object rather than going from layer to layer (Option variable layerhigh already unchecked).

The solution could be to separate printing the last layer per object, including ironing, and instead of that printing the last layer for all objects and do the ironing at the very end for the hole printbed.

Photos showing the process.

[image]
[image]
[image]
[image]

No 3MF file as am not allowed to share it.

bubnikv, 2020-12-07:
Why should ironing all the layers at once fix any under-extrusion issues? Due to differences in nozzle pressure?

As a workaround you may merge multiple objects into one (select them in plater, right mouse click -> select "merge" in the popup menu).

sbellon, 2021-01-08:
I even had the same issue with only on object, so I don't think the "merge" workaround will help in all cases.

Consider an object that has multiple planes on the topmost z layer (like e.g. an up-side down table with four legs). Currently (2.3.0-rc2) one top plane is extruded and then ironed, then the next top plane is extruded and ironed, etc. With that approach, the nozzle may already been clogged when extruding the topmost plane after an ironing step.

I also suggest that first all extrusion of one layer should be performed and then the ironing of that layer afterwards instead of interleaved.

barn53, 2021-05-10:
This approach here only helps with ironing topmost surface only.
The problems mentioned in other issues with ironing (#5113, #5634) would not benefit from this fix.
But nevertheless worth thinking of.

closed 2026-09-29 by prusaslicer-bot[bot]