prusa3d/PrusaSlicer #5001: Irrelvant option (No or Cancel) when using the Trashcan to Delete All

jweaver100, 2020-10-27:
**Version**
2.3.0-alpha2+win64

**Operating system type + version**
Windows 10

**3D printer brand / version + firmware version (if known)**
Prusa Mk3 and Mini

**Behavior**

This is a really minor issue... When you use the Trashcan/Delete All option, you get a popup which says "All Objects Will Be Removed" and givest you the option of Yes/No/Cancel

[image]

There is no reason to have No AND Cancel... It should be one or the other...

bubnikv, 2020-10-27:
I am not quite sure. "Cancel" to me is a panic button. I did something that
I want to run away from without thinking. That is what the "cancel" button
is for, I may click it without thinking.

ManuelMcLure, 2020-10-27:
In that case, I would remove the "No" button since it's redundant with "Cancel".

Area5142, 2020-10-28:
This is a standard dialog from the Windows GUI standard - Cancel is the same as pressing the X on the dialog box, while Yes and No is the actions. Cancel is part of a standard Windows dialog and in the tab order, where the X is not in the tab order.

In this case, pressing No and Cancel has the same effect, but as @bubnikv points out Cancel is the quick safe way to leave any Windows dialog without changes.

fe60, 2020-10-29:
Beside removing the "No" button I propose to change the label of the first button from "Yes" to "Delete all".
From my point of view this makes it easier to anticipate the actions triggered by the buttons.
[image]

Affected source code:
https://github.com/prusa3d/PrusaSlicer/blob/d053939994e99f9d156f45f1a1dce5b045e7618e/src/slic3r/GUI/Plater.cpp#L4779-L4782
Used library (not sure about the version):
https://docs.wxwidgets.org/3.1.0/classwx_message_dialog.html
Alternate solution:
`wxOK` instead of `wxYES_NO` and change the label of the Ok button
https://docs.wxwidgets.org/3.1.0/classwx_message_dialog.html#ad8fdf1426f7363bd1e1e07af5f1e7adc

Area5142, 2020-10-29:
@fe60 I think this would be a good solution.

lukasmatena, 2020-10-29:
I don't want to spoil anything, but do we even need the dialog now when we have undo/redo? I remember that the absence of undo/redo was the reason why it was introduced in the first place...

bubnikv, 2021-09-30:
Delete all confirmation is gone in PrusaSlicer 2.4.0-alpha2.

closed 2021-09-30 by bubnikv