prusa3d/PrusaSlicer #5005: Switching "Print Settings" from Saved project doesn't apply changes to selected preset

brettvitaz, 2020-10-27:
### Version
version_2.3.0-alpha2-37-g0798fa818

### Operating system type + version
Mac OS Catalina

### 3D printer brand / version + firmware version (if known)
MK3S + MMU2S

### Behavior
_Describe the problem_
Changed settings do no behave consistently when selecting new Print Settings after loading a saved 3mf file.

_Steps needed to reproduce the problem_
- Add object to build plate in PrusaSlicer
- Change infill % to 50%
- Save project
- Close PrusaSlicer
- Open PrusaSlicer
- Open saved project
- Change Print Settings to anything else

_Expected Results_
- Dialog appears asking to migrate changed print settings. e.g. [image]
- Infill % remains 50%

_Actual Results_
- Infill % is set to Print Settings profile default
[image]
[image]

#### Project File (.3MF) where problem occurs
[beefy_arm_final_config_test.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5448071/beefy_arm_final_config_test.3mf.zip)

lukasmatena, 2020-10-27:
I don't understand. If you load a 3MF, do not change anything and switch Print profile, there are no unsaved changes. Why should you get the 'Unsaved changes' dialog then? You can always get to the profile loaded from the 3MF simply by selecting it again.

bubnikv, 2020-12-07:
I believe we have fixed it for you with beta2. Would you please retest that it does what you expect?

bubnikv, 2020-12-07:
Namely, from the beta2 change log https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-beta2 :

When loading the presets from 3MF, AMF or G-code, PrusaSlicer used to create temporary "external" profiles. These "external" profiles were named with the name of the source file and suffixed with the name of the source profile in braces, for example "my_print.3mf (0.15mm OPTIMAL)". Starting with this release, the profiles loaded from 3MF, AMF or G-code referencing and modifying an existing local profile are no more loaded as "external", but the referenced profile is activated and modified with the loaded preset. If the referenced profile does not exist, but the profile refers to a system profile by the "inherits" field, the system profile is loaded and modified instead. This works for all profiles with the exception of multi-extruder printer with multiple filament profiles modified. In that case the first modified filament profile will be loaded as modified, while the other modified profiles will be loaded as "external".

rtyr, 2021-01-28:
Fixed in Beta2. No response. Closing.

closed 2021-01-28 by rtyr