prusa3d/PrusaSlicer #5006: Loading 3mf project with very small objects does not respect "convert to inches"

brettvitaz, 2020-10-27:
### Version
version_2.3.0-alpha2-37-g0798fa818

### Operating system type + version
Mac OS Catalina

### 3D printer brand / version + firmware version (if known)
MK3S + MMU2s

### Behavior
_Describe the problem_
Loading 3mf project with small object asks to "convert to inches" regardless of previously responding "No".

_Steps needed to reproduce the problem_
- Add very small object to build plate in PrusaSlicer
- Click "No" to "Saved in inches object detected" dialog
- Save project
- Reopen project

_Expected Results_
- File opens as saved

_Actual Results_
- "Saved in inches object detected" dialog appears every time project is opened.
[image]

#### Project File (.3MF) where problem occurs
[beefy_arm_final_inches_test.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5448111/beefy_arm_final_inches_test.3mf.zip)

brettvitaz, 2020-10-28:
Note: When saving a project where the object *was* converted to inches, subsequent loading of the project continues to load as inches.

bubnikv, 2020-12-07:
We should really NOT offer conversion from inches for 3MFs and AMFs, which are actually PrusaSlicer projects.

bubnikv, 2020-12-12:
Implemented in PrusaSlicer 2.3.0-beta3

closed 2020-12-12 by bubnikv