prusa3d/PrusaSlicer #5007: "Reload from disk" causes objects converted to inches to revert to mm

brettvitaz, 2020-10-28:
### Version
version_2.3.0-alpha2-37-g0798fa818

### Operating system type + version
Mac OS Catalina

### 3D printer brand / version + firmware version (if known)
MK3S + MMU2S

### Behavior
_Describe the problem_
When an object that is converted to inches is reloaded from disk, the object is reloaded in mm

_Steps needed to reproduce the problem_
- Add small object to build plate
- Answer Yes to convert to inches
- Right click on object, select "Reload from disk"

* _Expected Results_
- Object is reloaded at the converted size

* _Actual Results_
- Object is reloaded at mm size

#### Project File (.3MF) where problem occurs
[Single_Inch_Test.zip](https://github.com/prusa3d/PrusaSlicer/files/5448730/Single_Inch_Test.zip)

brettvitaz, 2020-12-04:
This is still an issue in Beta1

bubnikv, 2020-12-07:
You are right, we shall remember the "inches" flag and use it when reloading.

bubnikv, 2020-12-12:
Fixed in prusaSlicer 2.3.0-beta3

closed 2020-12-12 by bubnikv