prusa3d/PrusaSlicer #5009: Pressing Shift+Tab in prusa-gcodeviewer alpha2 shows PrusaSlicer sidebar

Area5142, 2020-10-28:
### Version
Symlink prusa-gcodeviewer to PrusaSlicer-2.3.0-alpha2+linux-x64-202010241644.AppImage

### Operating system type + version
Linux Mint 19.3 x64

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
Starting prusa-gcodeviewer from command line gives viewer and empty build plate:
[image]

Pressing Shift+Tab shows sidebar from PrusaSlicer:
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
No project/gcode needed

bubnikv, 2020-10-28:
That is intentional, it is even described in the alpha2 change log.

lukasmatena, 2020-10-28:
To clarify, it is not exactly intentional in gcode viewer mode. Fixed with 0a9f316.
Thanks, closing.

closed 2020-10-28 by lukasmatena