prusa3d/PrusaSlicer #5010: 2.3.0-alpha2 - Slicing result for 0.2mm differs from the results for all other layer heights

fe60, 2020-10-28:
### Version
2.3.0-alpha2

### Operating system type + version
Windows 10 (64 Bit)

### Behavior
Slicing my model with 0.2mm layer height generates a strange result. For about one third of a circular solid the last layer is missing.
[image]
It works fine with all other layer heights as well as with 0.2mm in Prusa Slicer 2.2.0.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[StrangeSlicing_3MFandSTL.zip](https://github.com/prusa3d/PrusaSlicer/files/5451613/StrangeSlicing_3MFandSTL.zip)

bubnikv, 2020-10-28:
https://github.com/prusa3d/PrusaSlicer/wiki/Frequently-Asked-Questions-(FAQ)#strange-pattern-on-a-flat-surface

fe60, 2020-10-28:
@bubnikv Thanks for this hint.
Generating the stl-file with the object already oriented upside down (flipped in Prusa Slicer before), solves my problem.
It's still strange, but I got a solution. Hence I close this ticket.

closed 2020-10-28 by fe60