prusa3d/PrusaSlicer #5011: Place on Face fails to get it flat

tslater2006, 2020-10-28:
### Version
Version 2.3.0-alpha1-win64

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
Ender 5 / Marlin 2.0 bugfix

### Behavior

STL is oriented veritcally by default:
[image]

Go to Place on Face and select the face with the red arrow:
[image]

Turn on brim to see what parts are touching build plate and slice:

[image]

PrusaSlicer has put this down at a slight angle. At first I thought it was the model having a curve but I tried the same steps in Cura and it places it completely flat on the build plate:
[image]

Ultimately used Chitubox's flatten by face and then exported the reoriented model as STL and imported that into PrusaSlicer:
[image]

_Is this a new feature request?_
No
#### Project File (.3MF) where problem occurs

[PlaceOnFaceBug.zip](https://github.com/prusa3d/PrusaSlicer/files/5452125/PlaceOnFaceBug.zip)

lukasmatena, 2020-10-28:
Thanks, reproduced. This seems to be same issue as #3505, which I briefly investigated and did not find the cause. I will get back to it, this is not cool at all.

fe60, 2020-11-06:
@tslater2006 
The issue #3505 mentioned by **lukasmatena** has been closed and the related fix is part of version [2.3.0-alpha3](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-alpha3).
I guess it's worth to give it a try.

lukasmatena, 2020-11-06:
@fe60 It turned out the fix was only enough to get rid of #3505. I know what is the problem with this one and I have a fix already, I'll commit it after I get to test it well enough.

lukasmatena, 2020-12-09:
It should be fixed with 2d6b694 for PrusaSlicer 2.3.0-beta3. Closing.

closed 2020-12-09 by lukasmatena

ShurikMur, 2021-02-08:
Hi, the same issue here.
Version 2.3.0 win64
The part is very simple with well-defined bottom face - lay flat function does not works.