prusa3d/PrusaSlicer #5012: Maxtemp runaway after first Layer

Dorito203, 2020-10-28:
### Version
2.3.0-alpha2+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
stock MK3s with Firmwareversion 3.9.0

### Behavior
The temperature of the hotend got out of control after the first layer was finished and the temperature should have been lowered.
At this moment the first line on the info display disappears, which contains the temperature display of the hotend and the Z-height. I can then go to the menu to set the temperature to the correct one, in places the display is buggy and the printer wants to heat up to almost 300 ° C. However, the automatic emergency shutdown works perfectly.

That was my first test print with the alpha2 version, unfortunately I have already closed the slicer and can no longer take screenshots of the settings that I have made. I could provide you with the gcode file if you need to.

_Is this a new feature request?_ No, but...

The monotonic infill leaves gaps in some smaller areas that should actually be filled.

rtyr, 2020-10-28:
> I could provide you with the gcode file if you need to.

Please provide the gcode.

Dorito203, 2020-10-28:
[0.2mm_PLA_MK3S_3h9m.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/5454555/0.2mm_PLA_MK3S_3h9m.gcode.zip)

rtyr, 2020-10-28:
Thanks, the issue is caused by a typo in the temperature setting.

Your filament settings:
[image]

Dorito203, 2020-10-28:
oh wow this is a little bit hot right there...
But it would be nice if the slicer had a limit or warn me in that case

rtyr, 2020-10-28:
Maximum allowed temperature is 1500C. It was 500C initially, but there was some demand for higher max temperature (printing various types of glass on special 3D printer).

This is mainly a task for printer firmware, but we may discuss this topic internally.

I believe we can close this issue.

closed 2020-10-29 by rtyr