prusa3d/PrusaSlicer #5013: Incorrect OpenGL warning when starting PrusaSlicer GCode Viewer with a file argument

VorpalBlade, 2020-10-28:
### Version
PrusaSlicer 2.3.0-alpha2+linux+x64 (GTK2). I'm using the AppImage. This issue does not happen with GTK3 build (but that one has broken UI, see issue #4911).

### Operating system type + version
_What OS are you using, and state any version #s_
Ubuntu Linux 18.04 x86-64
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

Here is the system info: [system_info.txt](https://github.com/prusa3d/PrusaSlicer/files/5453685/system_info.txt)

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Not applicable, but I have a stock Prusa i3 MK3s.

### Behavior
* _Describe the problem_

A popup with the following message is shown when starting `prusa-gcodeviewer` with a gcode file argument (but not when starting the normal slicer mode with a stl or 3mf as a file name argument or when starting the gcode viewer without arguments):
[image]

After this the program usually hangs. Sometimes it launches to an empty window instead though. About 1 times out of 10 it works as expected. This indicates to me that it is probably some sort of race condition or timing issue.

In all cases (even when it works, so probably unrelated), there are a lot of copies of the following two warning messages in the terminal:
```
(prusa-gcodeviewer:780): GLib-GObject-WARNING **: 18:06:21.372: invalid (NULL) pointer instance

(prusa-gcodeviewer:780): GLib-GObject-CRITICAL **: 18:06:21.372: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
```

Launching the program without a file argument and then opening the file always works.

* _Steps needed to reproduce the problem_
  * Launch `prusa-gcodeviewer` symlinked to `PrusaSlicer-2.3.0-alpha2+linux-x64-202010241644.AppImage` with a file name argument, e.g. (using some of the example gcode that came on the SD card of my MK3s): `./prusa-gcodeviewer Whistle_0.2mm_PLA_MK3S_34m.gcode`
* _Expected Results_: The gcode viewer opens and displays the file.
* _Actual Results_
  * The error message described above usually shows up, sometimes the program then hangs (and it is impossible to click OK), sometimes it continues opening and either shows a blank screen, blank build plate or just plain works.

_Is this a new feature request?_
No, it is a bug.

#### Project File (.3MF) where problem occurs
Happens with all gcode files I have tried. But here is one as an example:  [Whistle_0.2mm_PLA_MK3S_34m.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/5453691/Whistle_0.2mm_PLA_MK3S_34m.gcode.zip)

lukasmatena, 2020-10-28:
Thanks, we already noticed this. It happens when gcode viewer attemps to load a gcode file before opengl is initialized (through `GUI_App::post_init`). Otherwise the OpenGl is initialized during first render loop. Probably something for @enricoturri1966 to look at.

Area5142, 2020-10-29:
I reported a similar problem with symlink prusa-gcodeviewer on Linux Mint 19.3 in #4880, using AppImage.

lukasmatena, 2020-10-29:
@Area5142 Yes, thanks for the reference. These issues really seem connected.

closed 2020-11-03 by enricoturri1966