prusa3d/PrusaSlicer #5014: Program crashes after painting seam followed by adding raft

carloswaibl, 2020-10-28:
### Version
2.3.0-alpha2+

### Operating system type + version
macOS Catalina 10.15.7

### 3D printer brand / version + firmware version (if known)
Ender 3

### Behavior
The program quit unexpectedly while attempting to slice. 
I painted a seam using the new seam painter. Went to print settings -> Support material.
I then added a 2 layer raft (was 0 before). Went back to plater view, and clicked Slice now.
* _Expected Results_
Expected it to slice.
* _Actual Results_
Program quit while slicing. (crashed)

_Is this a new feature request?_
No

Not part specific. I tried with multiple parts following the steps outlined above.

lukasmatena, 2020-10-28:
Reproduced, thanks. It will be fixed.

lukasmatena, 2020-11-27:
Fixed with f485f66 for PrusaSlicer 2.3.0-beta1. Closing.

closed 2020-11-27 by lukasmatena