prusa3d/PrusaSlicer #5016: [2.3.0-alpha2] Support Interface layers in midair

vintagepc, 2020-10-28:
### Version
2.3.0-alpha2

### Operating system type + version
OpenSuSE 42.3 x64

### 3D printer brand / version + firmware version (if known)
MK3S//MMU2S, 3.9.1

### Behavior
When slicing the attached 3MF, I ended up with support interface layers suspended in midair, and nothing above them that requires support.

Also, #3061 is still present - the Interface layers setting is 2, but there are far more in many places. 

[image]

[image]

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs

[Midair_IF.3mf.gz](https://github.com/prusa3d/PrusaSlicer/files/5455335/Midair_IF.3mf.gz)

bubnikv, 2021-02-22:
Floating supports in PrusaSlicer 2.3.0
[image]

were fixed in current master
[image]

bubnikv, 2021-02-22:
Fixed with https://github.com/prusa3d/PrusaSlicer/commit/3985d7bb81ddd542612ec2e8704622be4e90f074

closed 2021-02-22 by bubnikv