prusa3d/PrusaSlicer #5018: Lost User Configurations

martinglass, 2020-10-29:
### Version
2.3.0 Alpha 2

### Operating system type + version
OSX (dmg from releases)

### Behavior
The new version deleted my user-saved configurations

* _Steps needed to reproduce the problem_
In the old version, save Configuration by selecting menu option "Take Configuration Snapshot"
Install 2.3.0 Alpha2 in place of previous 2.2 version

* _Expected Results_
Configuration Snapshots should show the User-saved Configuration

* _Actual Results_
The User-saved Configuration snapshot is deleted from the list.

neophyl, 2020-10-29:
When you say replace the previous 2.2 version with 2.3.0 Alpha do you mean replace the proper 2.2.0 release or one of the 2.2 alpha releases ?
The release notes state (and its been done this way for ages) that alpha versions of the software store their profiles in completely different location.  This is to make running both 2.2.0 and 2.3.0alpha completely separate.  The 2.2.0 profiles will still be there if you start up 2.2.0.  It means changes made when evaluating non finished software wont effect your 'production' set up at all.

If you want your 'proper' release profiles to be available in an alpha install you should open up the 2.2.0 release, and perform an export configuration bundle procedure.  That will export all your non system customised profiles to a file.  You then start up the alpha and do an import config bundle selecting the file you saved in the export.  That will copy them over between the release and alpha locations.
It does NOT copy the system ones so you would still need to run through the installation wizard to install any default printer/filament/print profiles.

lukasmatena, 2020-10-29:
Works as intended, closing. Thanks @neophyl for the explanation. The reason it is this way is to minimize risk that any bugs in the alpha destroy someone' cherished profiles.

closed 2020-10-29 by lukasmatena