prusa3d/PrusaSlicer #5019: Convert to inch question?

HaasNL2, 2020-10-29:
### Version
2.3.0 alpha2

### Operating system type + version
Windows 10

### Behavior
I am using Blender to make my models and I don't draw them in scale, I just scale them in the slicer. But now every-time I import a model of mine I get the inch warning. I sometimes print multiple version of the same model, they all need to be in the same scale. This inch conversion gives me 2 scales, one converted to inch and one default. This over complicates things for me!
I see there is no option in settings to disable this auto conversion, can there be added one?
[image]

fe60, 2020-11-13:
Duplicate of #5006.

HaasNL2, 2020-11-13:
Main issue I have is as it scales to inches it resets the scale to 1, while it is a scaled version of the model. 
So if you do automatic scale to inch at least adjust the scale accordingly don't just reset it to 1.

foreachthing, 2020-11-13:
Set the Units setting in Blender to **None**.
Then you can model in mm:

[image]

This will give you a cube of 20x20x20 mm in Prusa.

HaasNL2, 2020-11-20:
So the solution to this issue is not there should be an option to disable it but for me to change my default settings to something that is not correct but works in PrusaSlicer. That is not gonna happen!

Could someone please explain me why everyone needs this convert to inch option on by default? 
I can explain why we should not support something like inches. (makes angry science noises.) ;)

foreachthing, 2020-11-20:
It's not an imperial issue, the issue is that _you_ use the wrong units.
An STL has no units. So _you_ need to adjust your model accordingly.

BTW, you only need to do that once, in Blender, then save your startup file.

Inches are fun (let's you practice fractions ...) - I used to work with inches for a while and getting back to metric was harder than I thought. But those two or three countries still using inches are really stubborn! :-D

HaasNL2, 2020-11-26:
Inches are not fun and are totally retarded that make something simple a lot complicated it also waste a lot of resources that could be used for something useful. 
Please let me disable the inch stuff. If I change my default.blend I probably get somewhere else issues.

foreachthing, 2020-11-26:
> Inches are not fun and are totally retarded

True that! 

> If I change my default.blend I probably get somewhere else issues.

Just try it ;-)

bubnikv, 2020-12-07:
I believe this issue has been explained.
closing

closed 2020-12-07 by bubnikv