prusa3d/PrusaSlicer #5021: MMU2 MK3 initial tool load extruder speed wrong

NielschB, 2020-10-29:
When generating Gcode with use of MMU2 on MK3, the initial load speed of the extruder is set to 1073, while it should be 1140 to be in sync with MMU speed (19.01mm/sec)

rtyr, 2020-10-29:
Is this issue still valid? If so, please provide your gcode.

NielschB, 2020-10-31:
Not sure as my setup has changed to MK3S + MMU2S so the feeding overlap with MMU and Extruder is not present anymore.
Still, if the MMU2 feedrate is set to 19.01mm/sec, the extruder feedrate to match should be 1140 (1140 / 60 = 19) right?

rtyr, 2020-10-31:
Loading speed at the start is set to 19mm/s by default, so I don't see any issue at the moment. Closing.

[image]

closed 2020-10-31 by rtyr