prusa3d/PrusaSlicer #5022: Better way to handle nozzle move coordinates

gminadak, 2020-10-29:
From my understanding the slicer first sets the Z height and then moves to the position.
This results in combination with custom start code to raise the nozzle and instantly lower it on the exact same spot.
Is there anyway to change this behavior?

example:

;start gcode
**G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed**

; End of custom start G-Code
; Filament gcode

*G-CODE GENERATED BY SLICER*

G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
**G1 Z0.200 F9000.000**
G1 E-5.00000 F2700.00000
G92 E0
**G1 X73.765 Y96.222 F9000.000**
G1 E5.00000 F2700.00000**

daleckystepan, 2020-10-29:
If you are scratching bed at Z=0.2 something is probably wrong with your setup not in PS.

rtyr, 2021-02-01:
I am going to close this issue. It is already partially discussed here https://github.com/prusa3d/PrusaSlicer/issues/2420.

closed 2021-02-01 by rtyr