prusa3d/PrusaSlicer #5023: [Request] Define location for user saved presets

mvasilakis, 2020-10-29:
It would be nice to be able to define the save location for user saved presets. For OS X they are saved in /users/user/Library/Application Support/Prusa Slicer/(print, printer, filament)

While it's fairly trivial to create a hard link and place the directory where I'd like it might be useful for others to have the ability to define this location. My reason for this is so that I can define a dropbox directory where my profiles reside. This way I have access to them on any computer i use and they are the latest saved versions.

Le0Michine, 2021-07-26:
Hi, just wanted to reiterate on this feature request. Is there any chance it gets implemented?

I believe the option with symlinks on macos/linux and --datadir is a bit tedious for an application which has a GUI. Because it isn't exactly an official way to go the symlink might get overwritten with an update. On windows I use multiple ways to start the app (for example fusion360, start menu, taskbar, some other 3d modelling software) so making sure they all use the same shortcut is also suboptimal. It would be great to see this in one of the future releases as it seems like a great UX improvement for quite a few users.

eneene, 2022-03-03:
The slicer itself could have a dropbox/icloud/google drive app. You link prusaslicer to a cloud service. I know it is more than a little feature, but the value to the end user is huge.
It could go even further than simply storing the presets to become a proper file manager.

eneene, 2022-08-13:
Just because we have a new version coming out, i'd like to mention this feature again. At least let us set the config files folder. I would point it to a dropbox folder and life would be a lot easier.
I slice on my laptop away from the printers a lot!
Thank you so much.

alpen-werk, 2023-08-23:
> 1. Create shortcut
> 2. Open shortcut properties, add the --datadir parameter
> 
> [image]

Works great ! Thanks

Sembazuru, 2026-09-03:
the --datadir flag points to the location where the PrusaSlicer.ini and other files are. It isn't the program settings that I want shared across different computers, but the user profiles and user gallery shapes (and now user plugins) that I want shared across different computers. Please, with the new version, include a setting that is saved in PrusaSlicer.ini to point to a different path where the user profiles, shapes, and plugins are stored to allow us to point that to a cloud service synchronized service folder. My current method of using links to point the shared folders takes more filesystem knowledge to set up than should be expected from the general user.

Jan-Soustruznik, 2026-09-08:
Hello, @Sembazuru , thank you for update, According to new [policy](https://github.com/prusa3d/PrusaSlicer?tab=contributing-ov-file#issue-tracker-policy) I closing this feature request and converting it into a Discussion thread, as I'd be interested to hear what the community thinks about how it should work. Thank you, and have a nice day.