prusa3d/PrusaSlicer #5024: Backwards Compatibility Issue

GWdd, 2020-10-29:
### Version
2.3.0 Alpha 2

### Operating system type + version
Windows 10 2004

### 3D printer brand / version + firmware version (if known)
Prusa Mk3

### Behavior
Open an stl in v2.3.0a2 - slice, save as .3mf project.
Attempt to open in 2.2.0, flags error.
[image]

#### Project File (.3MF) where problem occurs
2.3.0a2 project attached:
[BkwdsCompat.zip](https://github.com/prusa3d/PrusaSlicer/files/5460489/BkwdsCompat.zip)

Honestly, I'm not sure this is a major issue - but always good to check for those that may share files with folk who haven't yet upgraded.

bubnikv, 2020-12-08:
This is an unfortunate situation and we cannot do much about it, we cannot revert time and change 2.2.0 release.
We are striving to make PrusaSlicer backward compatible by adding new configuration fields in a way, that if the old slicer just ignores them, it will still produce a somehow usable result.

Unfortunately PrusaSlicer 2.3.0 introduces a new Monotonic infill type and PrusaSlicer 2.2.0 does not like the keyword and bails out with a nonspecific error message: "ConfigBase::set_deserialize() failed".

PrusaSlicer 2.3.0-beta2 already produces a somehow more useful error report.
[image]

We should think hard now before 2.3.0 release a show to make 2.3.0 resilient against breaking changes in 2.4.0, but as of now I have no idea how to do that. We can certainly introduce config versions, so 2.3.0 will refuse a config or a 3MF containing incompatible config, but that would not be much different from what we currently have. We could likely report these offending values and revert them to their defaults, however that would take some time to implement correctly.

bubnikv, 2020-12-08:
For reference, here is an example of an incompatible (broken) 3MF with a phony "monotonious" infill type.
[Shape-Box-incompatible.zip](https://github.com/prusa3d/PrusaSlicer/files/5658573/Shape-Box-incompatible.zip)

GWdd, 2020-12-08:
New features are hard to plan for ... we agree.

Perhaps changing the error handler in a future version so that these errors are flagged as warnings, not fatal errors. Probably a lot more work, but adding a fall back default for settings fill when a type is unknown. Then report "Warning: unknown bottom_fill_pattern 'monotonic' , using 'rectilinear'." ... 

Or an easier/simpler approach - flag the .3MF file version as the reason there is or might be errors. By adding a Prusa specific version number in the metadata any checking and flagging "unknown" compatibility becomes easy. 

Or do nothing. If I'm the only one even concerned it isn't worth worrying about.

JohnOCFII, 2020-12-09:
It might be helpful to let users know (whether here in the error message, or some linked document) that the core model/STL itself is OK, it is just the additional information in the 3MF causing an issue.

bubnikv, 2020-12-09:
We should do something. Unfortunately it is not a one linear and we will
most likely not squeeze it into 2.3.0 final release, thus the fix we will
do in 2.4.0 release will help to open projects from 2.5.0 release. Bummer.

bubnikv, 2021-09-02:
Implemented in PrusaSlicer 2.3.3.
User is given a "substitution" dialog for enums that are not understood by an older slicer. Also replacing booleans with enums is supported by the "substitution" dialog, thus we may extend a boolean in the future with multiple enumerated values.

closed 2021-09-02 by bubnikv