prusa3d/PrusaSlicer #5031: Prusaslicer creating gaps between tree support and model.

Devenus3, 2020-10-30:
### Version
Version 2.2.0+win64

### Operating system type + version
Windows 10 Version 2004 (OS Build 19041.572)

### 3D printer brand / version + firmware version (if known)
Prusa MK3S Kit Build, Firmware 3.9.0-3421

### Behavior
* _Describe the problem_

So, I have a model that started like this:

[image]

I added supports using the FDM printer settings and exported them
So the model looked like this when I went to slice:

[image]

However, when I do slice it, I end up with these gaps:

[image]

[image]

I've been tweaking some settings to deal with oozing, to an amount of success, but I noticed this problem happening even when I revert to default settings.

This has also been occurring on other models with similar supports generated via prusa slicer.

Tweaking layer height still causes the same gaps.

This issue doesn't seem to be present when using the presets of The Prusa Mk3s printer with a .25 nozzle diameter. 

[image]

The initial screenshots were based on a profile that used the Prusa Mk3s Printer with .4 nozzle diameter as a base.

This issue seems to be occurring on every model I've tried slicing with similar supports, which again, were achieved by switching to the Prusa SL1 profile, generating supports, and exporting the model with supports as a single stl. 

(To note: The above model has been printed just fine in the past without issue, when using a .25 nozzle with the .25 settings. This issue only became apparent after switching back to the .4 nozzle and using the printer settings in prusaslicer.  The slicer slices it cleanly when using the default .25 settings.)

_Is this a new feature request?_

No

#### Project File (.3MF) where problem occurs

[Treesupportissue.zip](https://github.com/prusa3d/PrusaSlicer/files/5465380/Treesupportissue.zip)

neophyl, 2020-10-30:
The contact points between the supports and the model are obviously smaller than is printable with the 0.4 nozzle so it doesn't slice them.  Whats so unusual about that ?  That's normal operation.

Devenus3, 2020-10-30:
> The contact points between the supports and the model are obviously smaller than is printable with the 0.4 nozzle so it doesn't slice them. Whats so unusual about that ? That's normal operation.

I was confused by it and reached out to the chat support on the prusa website.  The rep there told me that it was a bug and to post about it here. How can you tell that the supports are smaller than is printable?

rtyr, 2020-10-30:
@neophyl is right. Not surprising result considering these supports are not designed for FDM printing. If you need this type of the supports for FDM, use Meshmixer supports instead https://blog.prusaprinters.org/how-to-create-custom-overhang-supports-in-meshmixer_8188/.

Closing, I don't see any issue here.

closed 2020-10-30 by rtyr