prusa3d/PrusaSlicer #5035: Cannot Drag and Drop STL's into slicer

Cosantoir, 2020-10-30:
### Version
2.3.0-alpha2+win64

### Operating system type + version
Windows 10
### 3D printer brand / version + firmware version (if known)
NA

### Behavior
* When downloading STL's they cannot be dragged and dropped onto the slicer window to open the STL
* Open PrusaSlicer Aloha2, Download STL, open Download folder, drag and drop STL onto prusa slicer window
* Expect that the file should open and render on the print bed of prusa slicer
* Actual result, nothing happens

#### Project File (.3MF) where problem occurs
NA

GWdd, 2020-10-30:
I drag and drop files in all the time ... 2.3.0a2 + win64pro 2004  ...  and just did it one more time to be sure. Works as expected.

bubnikv, 2020-12-07:
@Cosantoir 
Do you still have this issue with beta2?
It is really weird, that you are the only one reporting this issue.

rtyr, 2021-01-28:
No response. Closing.

closed 2021-01-28 by rtyr