prusa3d/PrusaSlicer #5036: NFR | Is it possible to apply monotonic infill into first layer?

MrTantrum, 2020-10-30:
I've been working with Prusa Slicer 3.0 Alpha 2, and am really liking the new monotonic infill for the top layers.  However, I often print objects where the bottom layer is actually the top of the model when displayed/utilized.  Given that the first layer reflection has the same issues as the last layer does when infill is applied in multiple directions, has it been considered to also automatically apply or to provide the option for monotonic infill for the first/initial layer?

rtyr, 2020-10-30:
It is already possible.
[image]

MrTantrum, 2020-10-30:
That is great, sorry I overlooked it.  Thank you for taking the time to point out where the setting is.

closed 2020-10-30 by MrTantrum