prusa3d/PrusaSlicer #5037: Windows UI Anomaly - PS flashing all desktop icons at startup

GWdd, 2020-10-30:
### Version
2.3.0a2

### Operating system type + version
Windows 10 pro 2004 / 20H2
Possible hardware relation: nVidia RTX driving 3840x2160 (4k) display

### 3D printer brand / version + firmware version (if known)
Prusa Mk3

### Behavior
When starting Prusa Slicer, the Windows UI flashes all of the icons, as if the entire icon library is being updated. Indeed, when I have been able to 'freeze' the screen during this flash of change, all of the desktop icons are being changed to system defaults then back to registered apps.

* _Steps needed to reproduce the problem_
Simply start Prusa Slicer by double clicking a shortcut or vua Start>PrusaSlicer link.

Apps generally don't need to refresh the icon cache every time they start. Ill-behaved app are know to do this, but reasonable apps don't. 

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
Not something a screenshot will show.

#### Project File (.3MF) where problem occurs
No project needed to demonstrate the issue - just start the app.

Tried recording the event but the simpler recording tools I have won't grab the entire desktop.

andrewluebke, 2020-10-30:
This doesn't happen for me.  I'm running on a similar setup with a RTX2080Ti, driver version I'm using is 451.82 if that helps you.

GWdd, 2020-10-30:
I'm on 451.67 ... doubt it is an nVidia issue. It's been happening almost since day one, even on Windows 7.  I have seen oddities with other apps and 4k and interactions with DirectX.  But it is only PS that does the icon replace every start up.

bubnikv, 2020-11-02:
I think the file types are re-registered at file explorer at every prusaslicer start.
@enricoturri1966 would you please look into this?

enricoturri1966, 2020-11-05:
@GWdd
This should happen only the first time you run PrusaSlicer.
Are you running different versions of PrusaSlicer from different folders ?
If so, this will happens every time you change the exe you are using because of icon association to the selected exe

GWdd, 2020-11-05:
@enricoturri1966 
Thank you - that almost makes sense; there are times where the icons do not flicker and I am using the stock 2.2.0 and 2.3.0 interchangeable. It also explains why the effect is persistent over time - even before 2.3.0a.

closed 2020-11-05 by GWdd