prusa3d/PrusaSlicer #5039: Build libslic3r as dll or Visual Studio project

AndreasReitberger, 2020-10-31:
### Offtopic, not an issue
I'm trying to build a dll out of the slicer library, however I don't get it done using cmake.
It always says it cannot find "PrecompiledHeaders".
Has anybody made a dll out if the lib? Or better a Visual Studio project?

...src\libslic3r

It would be awesome if anybody could share the created Visual Studio project here.
Thanks!

foreachthing, 2020-10-31:
This has always been my reference: 
https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Windows.md#noob-guide-step-by-step

AndreasReitberger, 2020-10-31:
> This has always been my reference: 
> 
> https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Windows.md#noob-guide-step-by-step
> 
> 

Yes, this was the first thing I've tried.. however I always get an error that the "PrecompiledHeaders" cannot be found.

foreachthing, 2020-10-31:
I've made this batch file and this works like a charm.
This batch assumes VS2019.

	@echo off

	ECHO I'm going to CLONE the PrusaSlicer repository - ready?
	ECHO Best, if you're curently in c:\dev (or similar short path)
	ECHO Also make sure you got at least 15 GB of free disk space available
	ECHO.
	ECHO Press CTRL+BREAK if you want to cancel.
	PAUSE
	ECHO Are you sure??? Shall I continue?
	PAUSE

	git clone https://github.com/prusa3d/PrusaSlicer.git --recursive
	cd PrusaSlicer

	SET "thisPATH=%CD%"
	echo Building in %thisPATH%

	cd deps
	mkdir build
	cd build

	cmake .. -G "Visual Studio 16 2019" -DDESTDIR="%thisPATH%"

	REM This will take a loooong time! Go get some coffee...
	msbuild /m ALL_BUILD.vcxproj

	cd "%thisPATH%"
	mkdir build
	cd build
	cmake .. -G "Visual Studio 16 2019" -DCMAKE_PREFIX_PATH="%thisPATH%\usr\local" -Wno-dev -Wno-error=deprecated

	msbuild -ds -nologo PrusaSlicer.sln -t:rebuild -m -p:Configuration=Release -verbosity:normal

    REM Robocopy to the release dir:
    REM Make sure this destination dir exists.
    rem robocopy "c:\dev\PrusaSlicer\build\src\Release" "c:\PrusaSlicer-Release" /MIR /xf *.exp *.lib *.pdb /R:600 /W:30

You'll need to compile it yourself. This directory requires ~12 GB, so I can't really share it ...

foreachthing, 2020-10-31:
Then, if I want to rebuild PS, I just start this PowerShell-Script (in `c:\dev\PrusaSlicer\`):

    $gitstat=git pull --verbose
    $gitcommit=git rev-parse HEAD
    $done=0

    function RebuildPrusaSlicer {
        cd build
        msbuild -ds -nologo PrusaSlicer.sln -t:rebuild -m -p:Configuration=Release -verbosity:normal
        cd ..
    }

    function SyncRelease {
        Write-Output "====== Mirror to PrusaSlicer-Release ======"
        robocopy "c:\dev\PrusaSlicer\build\src\Release" "c:\PrusaSlicer-Release" /MIR /xf *.exp *.lib *.pdb /R:600 /W:30
    }

    If (($gitstat -join '') -eq ('Already up to date.')) { 
        Write-Output ""
        Write-Output "### Already up to date."
        Write-Output ""
            
        $confirmation = Read-Host "Do you want to copy the current build anyway?" `
            "(y/n) Or do you want to force rebuild and sync? (R)"
        if ($confirmation -eq 'y') {
            SyncRelease
            $done=1
        }	
        Elseif ($confirmation -eq 'R') {
            RebuildPrusaSlicer	
            SyncRelease
            $done=1
        } else {
            $done=0
        }

    } Else { 

        Write-Output ""
        Write-Output "====== Rebuilding now. This will take some time! ======"
        
        Start-Sleep -s 2

        RebuildPrusaSlicer
        
        SyncRelease
        
        $done=1

    }

    if ($done -eq 1){
        Write-Output ""
        Write-Output $("Commit now at : " + $gitcommit.SubString(0,7))	
        Write-Output ""
    } else {
        Write-Output ""
        Write-Output "Build and sync aborted"	
        Write-Output ""
    }

Good luck!

AndreasReitberger, 2020-11-01:
@foreachthing 
Awesome! Your script is working fine on my end. Thanks for sharing!

closed 2020-11-01 by AndreasReitberger

foreachthing, 2020-11-02:
@AndreasReitberger, I'm glad I could help.
Now ... do you know what you might have done different from my script? Basically, my script does what the "noob guide" says. I'd like to know, because if the "noob guide" was unclear, I might consider a PR on the guide and add my script to it.

AndreasReitberger, 2020-11-02:
Hi @foreachthing 
actually the guide is fine. First the batch wasn't working either. I had to re-install cmake manually. Then it started to work with your batch. Guess there were some problems.
Thanks for helping out!

sorrowfeng, 2020-11-07:
When I compile the last step, I get more than a thousand errors for the project libslic3r_gui, what is wrong with it?

bubnikv, 2021-08-06:
# [image] PrusaSlicer 2.4.0-alpha1

## Summary

This is the first alpha release of PrusaSlicer 2.4.0, introducing ..., ... new 3rd party printer profiles and many more new features, improvements and bugfixes.

To let you enjoy the alpha without worries, the alpha builds save their profiles into PrusaSlicer-alpha directory, so you may use the alpha side by side with the current release without ruining your production configuration.

Complete description of improvements and changes with respect to the current [PrusaSlicer 2.3.2 release](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.2) follows.

## Multi material segmentation by painting #3535

In PrusaSlicer 2.3.0 we have introduced paint-on supports and seams. Seeing the painting tools in PrusaSlicer, many customers asked for similar tools to help with multi-materal printing. Indeed, preparing models for multi-material FDM prints used to be a laborious and time consuming process. With this release of PrusaSlicer, it is no more and coloring an object for multi-material printing requires just a handful of bucket fill and brush stroke operations.

The editing tool offers the following tools:
1) Painting tool is similar to the old support or seam painting tool. 
2) Segmentation tool seed fills a region below the mouse cursor up to sharp edge, where the sharp edge threshold is adjustable at the gizmo panel.
3) Bucket fill tool to replace a patch of one color with another color.

Lze si pro levé a pravé tlačítko myši zvolit konkrétní extruder. Při vybírání extrůderů je zobrazeno jeho jméno (číslo) a barva. Defaultně se kreslí tou barvou, kterou má extruder, ale vedle comboboxu to jde změnit na libovolnou. Takhle změněná barva má trvání do zavření a do 3MF souboru se neukládají, potom jsou načteny výchozí barvy extrůderů.
Shift + Levé tlačítko myši, nezávisle na volbě extruderu nastavuje defaultní barvu.
Seed fill tam je taky změněný, teďka už nemá tak kostrbaté ovládání. Jen se vypíná/zapíná a mění úhel. Zatím je vizualizovaný fixní zelenou barvou a aplikuje se až kliknutím. Podle tlačítka myši se aplikuje barva.
mmu_segmented_region_max_width

Once the object is painted, the object surface is segmented into patches of same material and the patches are extended inwards. The regions painted at the top resp. bottom surfaces are extended by the number of "top_solid_layers" resp. "bottom_solid_layers", while the regions painted at the side walls are expanded inwards up to maximum "mmu_segmented_region_max_width".

Please note that thin features are not printable with FDM technolgy, thus for example painting a thin wall with two distinct colors will not make the thin wall printable. Such thin non-printable regions created during segmentation are thus detected and merged with the neighboring region, changing its color. PrusaSlicer will print thinner features with "detect thin walls" enabled, thus one is advised to try to enable "detect thin walls" if painting over thin features.

## FDM support improvements

FDM supports were redesigned:

1) PrusaSlicer newly offers printing bridges and 1st object layers over supports with the default layer thickness. The bridging behavior is configurable with a new "thick_bridges" boolean option. If enabled, the old Slic3r bridging strategy is applied, while if disabled, bridging is performed in a similar fashion to the other FDM slicers, leading to better bottom quality over supports. With "thick_bridges" disabled, the behavior of "bridge_flow_ratio" changes: Instead of modifying cross section of a rounded extrusion while keeping the gap between the extrusion threads constant, extrusion spacing is maintained while modifying the extrusion width and / or height to reach the prescribed extrusion ratio.

2) New "snug" shape of the support columns. By default, PrusaSlicer projects support columns into a regular grid for stability. Supports projected into a regular grid have a well defined minimum area, however they tend to leak out of regions they support along the object walls down to the print bed. We newly implement "snug" supports, which maintain the shape of the supported overhangs, thus avoiding the leaks with the trade-off of possibly lower support stability #555 #740 #902 #1346 #1993 #2708 #3083 #4783 #6240.

The support shape is configured with a new "support_material_style" config field, which accepts two values: "grid" (the old behavior) vs. "snug" (the new shape). The "snug" supports accept a new parameter "support_material_closing_radius", which controls merging close support islands and simplification of the support column shape. We borrowed a function smooth_outward() from Cura written by @BagelOrb to robustly simplify support island contours during merging of multiple support overhang islands.

3) If "support on build plate only" is enabled, the support columns are newly trimmed to not land on top of an object. However this may make the column too small to be stable #915 #1340 #3980 #4199.

4) Support enforcers newly take precedence over "supports on build plate only" and over "don't support bridges". This applies to both the enforcer and blocker modifier meshes and paint-on supports. Also the fact that the enforcers take precedence over blockers is indicated by sorting the enforcer meshes below the blocker modifiers at the side panel (the volume lower in the list always takes precedence over a volume upper in the list) #1870 #1997 #2801 #3741 #3831 #5105 #5748.

[... cut ...]
                 Delete part without settings => Single part object without settings, but settings are applied for the object.

//FIXME ExPolygon::overlaps() shall be commutative, it is not!
 
e002f0066fd96f38a5241a429f2228278de28f1d

transparent modifiers

Enrico why?
Modified version of automatic downscale on load of objects too big 526233ca472da33070a0d576fe8144100e02cb43 1e69e6def14940e044d053d3baa55075302d2f2f

enrico GCodeProcessor use new flavor gcfMarlinFirmware 52af90ed8f9f8ab2c8a306da42c3909e133dba88 9b2d9fb01a9f5189041c07fba8a419ad424e3c0b review?

CGAL corefine, Filip CGAL boolean operatory!