prusa3d/PrusaSlicer #5040: Feature Request: Use the new notification system when uploading gcode to printer

Area5142, 2020-10-31:
### Version
PrusaSlicer-2.3.0-alpha2, latest and previous alphas

### Operating system type + version
All

### 3D printer brand / version + firmware version (if known)
All

### Behavior
When using gcode upload to Toshiba FlashAir:
[image]

there is a notification:
[image]

Instead of that Windows frame notification it would be better to have the notification in the new pop-up area.

And a similar notification when the upload has finished (is missing today).
On slow Wi-Fi connections and with large models the only way to see if the upload has finished is to use the job-list (Ctrl+J) - a simple notification like "All uploads has finished" would be better.

_Is this a new feature request?_
Yes

#### Project File (.3MF) where problem occurs
No project/gcode needed

bubnikv, 2020-12-07:
That certainly makes sense, but we will certainly not squeeze into PrusaSlicer 2.3 release.

rtyr, 2021-08-25:
Implemented on master. It will be included in the upcoming PrusaSlicer 2.4.0-alplha1. Closing.

[image]

closed 2021-08-25 by rtyr