prusa3d/PrusaSlicer #5043: Print quality decreases when switching from gap fill to perimeter

NielschB, 2020-10-31:
### Version
2.2.0+win64

### Operating system type + version
Windows 10 pro 64bit 

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S+MMU2S FW 3.9.1

### Behavior
When slicing my model (cilinder) up to a certain height prusaslicer shows outer perimeter+gapfill+inner perimeter, then next layer it switches to outper perimeter+perimeter+inner perimeter. This is rendered fine, but when printed, the last part shows a significant decrease in print quality as shown below. (top part)
[image]

prusaslicer layer before the issue:
[image]

prusaslicer layer when issue starts:
[image]

#### Project File (.3MF) where problem occurs
[Ledring_lamp_tube.zip](https://github.com/prusa3d/PrusaSlicer/files/5469652/Ledring_lamp_tube.zip)

bubnikv, 2020-12-09:
Enabling "detect thin walls" should help in this case.

NielschB, 2020-12-16:
I'll try this option. Is it expected to solve the bad print quality problem on the top part? As I see it this should just not happen, regardless of the "detect thin walls" setting.

NielschB, 2020-12-23:
When using the "detect thin walls" option the quality problem is indeed solved, can be used as workaround for the moment.
Thank you!

closed 2026-09-29 by prusaslicer-bot[bot]