prusa3d/PrusaSlicer #5044: cmd + shft + q doesn't send to octoprint

3Dietrich, 2020-10-31:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

YuSanka, 2020-12-04:
Thanks for your report. But since we cannot reproduce your error, we have an assumption that you are observing. We came up with 2 options:
1. You have 2 displays and the app is on a secondary display. But the PrintHostDialog is always showing on the primary display (this was a bug fixed with https://github.com/prusa3d/PrusaSlicer/commit/e40ed644c893cc73373d83d4139f25182b7cc036)
2. You have no print host information. To do this, you must select a physical printer, and its "Hostname, IP or URL" parameter must not be empty. The "Send G-code" button can be used as a control point.
Do you notice this button? 
[image]

3Dietrich, 2020-12-04:
Now I can't reproduce myself anymore!
However, **none of your options** are true.
I remember I had an error message down right, but when I forced one, it still worked..
...I don't know...sorry!

So I suggest to close this and write another bugreport when I have a clean repro

YuSanka, 2020-12-04:
Ok, thanks. I close this issue :)

closed 2020-12-04 by YuSanka