prusa3d/PrusaSlicer #5045: "ironing" is hard to find

3Dietrich, 2020-10-31:
### Version
2.3alpha2
### Operating system type + version
MacOS Catalina 10.15.7

### 3D printer brand / version + firmware version (if known)
i3MK3S

### Behavior
I was very curious about the "Ironing" feature. But i couldn't find anywhere how to call it up. i read git prusa alpha3 - nothing, alpha2 - nothing! prusaprinters.org - nothing! where on earth can i find the feature!

I could only find a mini small entry in the 2nd submenu of the right mouse button on the object - is that it? Is the new feature really hidden there?

**Steps needed to reproduce the problem**
I found this "ironing" in the 2.step of right mouse popup after some search but I **forgot** about this because it's a modifier and not a "new main feature". I searched and searched... no way..
**Expected Results**
place Ironing in Print Settings or anywhere else where it can be honored as a print feature!
**Actual Results**
* right click on the object
* in the pop-up menu search (the last entry, sometimes unvisible so you have to scroll...) "Add settings"
* click this - you are now in the submenu of the rehcte mouse popup menu and finally find "Ironing". 
* Click this and go to the next level of settings. 
* there comes a (too small) window ..activate all
* and then (in the sidebar) activate all again..%$#)
_Is this a new feature request?_
well.. if you think how I just described workflow for getting ironing is good - then: YES.

rtyr, 2020-10-31:
I don't think it is so hard to find :-).
[image]

3Dietrich, 2020-11-01:
> I don't think it is so hard to find :-) (make sure you are not in the simple mode).
> [image]

krass! I really searched.. and didn't find this.. damn!! Yea, this is correct indeed .

3Dietrich, 2020-11-01:
close quietly

closed 2020-11-01 by 3Dietrich

rtyr, 2020-11-01:
I believe we can close this.

bubnikv, 2020-11-02:
Also the new "fuzzy search" helps a lot.

3Dietrich, 2020-11-02:
yes! oh man.. i was sitting on my brain and eyes at the same time..
The whole issue is very embarrassing for me