prusa3d/PrusaSlicer #5046: ironing doesn't work as modifier

3Dietrich, 2020-10-31:
### Version
2.3alpha2

### Operating system type + version
MacOS Catalina 10.15.7

### 3D printer brand / version + firmware version (if known)
i3MK3S

### Behavior
I don't want to iron every top service (but also not only the top most), so I used ironing as a modifier
** Steps needed to reproduce**
* in editor view, at the print object right mouse - Add modifier - Cylinder
* place it correctly (see attachment)
* Add Settings -->Ironing to the new object in sidebar (named "Generic-Cylinder") and activate everything in the (too small) window
* in sidebar activate Ironing and set Ironing Type "Topmost surface"

**Expected Results**
it irons the topmost surface inside the Generic Cylinder
**Actual Results**
it does nothing

* _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]
[image]

_Is this a new feature request?_
no, but a bug
#### Project File (.3MF) where problem occurs
[Jogurt_F.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5469937/Jogurt_F.3mf.zip)

Matts-Hub, 2020-11-07:
I think that this is the expected result. You are telling the slicer that in this area of the part (the modifier), to only iron on the topmost surface of the part. This is not the same thing as only ironing the topmost surface *of the area within the modifier*, which I don't think is a possibility.

Basically, where you place the modifier does not change what the slicer considers to be a topmost surface - that is defined by the entire part and not the area within the modifier. 

For your situation, if you only want that bottom circular area to be ironed and not the "F", you can move the modifier upwards so that layer isn't contained in it, and then set the ironing type to all top surfaces.

3Dietrich, 2020-11-07:
I see your point after you explained it to me. But "expected"? For many users it will be difficult to understand that the " top most surface" of a complete object is meant in a special modifier that does not include the top most surface.

thank you for the tip, it's a simple but good trick

closed 2026-09-29 by prusaslicer-bot[bot]