prusa3d/PrusaSlicer #5047: 2-finger scroll (normally zooming) doesn't work every where

3Dietrich, 2020-10-31:
### Version
2.3alpha2

### Operating system type + version
MacOS Catalina 10.15.7

### 3D printer brand / version + firmware version (if known)
i3MK3S

### Behavior
**Describe the problem**
on apple macbook trackpad, sliding with 2 fingers usually zoom in/out. In the alpha it sometimes it just does not work..

**_Steps needed to reproduce_**
* in Preview
* with "Legend" open
* slide 2 fingers in the upper left 

**_Expected Results_**
it should allways zoom in/out
**_Actual Results_**
it doesnt zoom
sometimes it selects something it the Legend! 
(I geus legend area is anyhow doublesized..)

in this (rough) area 2-finger slide doesn't zoom
[image]

_Is this a new feature request?_
nope

rtyr, 2020-10-31:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/4981?

3Dietrich, 2020-11-01:
> Duplicate of #4981?

yep, seems so

closed 2020-11-01 by rtyr